FreeCAD/FreeCAD #5903: Autofill information in SVG page template

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3911

* **Reporter:** vocx
* **Date submitted:** 3/20/2019
* **FreeCAD version:** 0.18
* **Category:** Feature
* **Status:** new
* **Tags:** SVG, TechDraw

# Original report text

In Catia the Drawing workbench auto-fills certain text fields in the Page Templates that come with the software. TechDraw could implement a similar behavior.

The usage would be something like a button in the TechDraw toolbar to open a dialog with the fields that can be filled in the page template. It could also have a button in this dialog to automatically fill at least some values like Author, Date, Title, and Page. Another option would be to right click the field in the page itself (highlighted in green) to display the options (Author, Title, Page, etc.) which will to fill the field.

In order to implement this well, each SVG template provided by FreeCAD should conform to some standardized name for the editable fields, for example, FC_AUTHOR, FC_DATE, FC_TITLE, etc. The data to write in the fields can be taken from the General Preferences of the Document, or taken from a Spreadsheet, or calculated by Qt in some way.

The discussion is on the forum https://forum.freecadweb.org/viewtopic.php?f=35&t=31681

# Additional information

OS: Ubuntu 18.04.2 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16093 (Git)
Build type: Release
Branch: releases/FreeCAD-0-18
Hash: 690774c0effe4fd7b8d2b5e2fb2b8c8d145e21ce
Python version: 2.7.15rc1
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by wandererfan 2020-01-05 12:46
[additional forum discussion](https://forum.freecadweb.org/viewtopic.php?f=8&t=42181)

---

### Comment by svbol 2020-01-05 20:53
I think - this small but very need function.
I add screenshots about this feature.

May be - this order:
1. Create list of standard field's names,
2. Add option in TechDraw setting - "Insert information for page",
3. Change all template (rename all fields),

titanic-jb, 2022-06-26:
I agree, that this is a useful feature.  Compare for example qelectrotech, which uses %-variables.  There is a bunch of variables, which are defined by the software.  Some of them are project related, some are related to the actual sheet.  The sheet variable will override the project variables.

The user can define new variables and use them also in the field of the drawing.  For example, and automatic date field, updated whenever the file is saved, is inserted by %{date}.  Therefore the user can choose, if the date (or any other) field should automatically be changed or not.

I'd suggest, that you add such a properties section to the project, this could be very helpful.

WandererFan, 2022-06-26:
@luzpaz : if I search with "is:issue is:open label:"WB TechDraw" ", this one isn't in the list.  I can only find it by entering the issue number.  Does it need a tag added or ????

luzpaz, 2022-06-27:
@WandererFan  see screenshot: 
[image]  
No problem for me

maxwxyz, 2024-03-02:
https://github.com/FreeCAD/FreeCAD/pull/11682 partially solved this.
Remarks from the FreeCAD Day 2024 Complaint Session:
When using a template in tech draw certain things like scaling is not automatically put in the title block. You set the scale and then you have to to type in the scale again. You you have to type in your name and other fields do not get automatically populated in the fields that visually represent that.

nm2107, 2024-03-04:
Hello guys :)

Such feature would be very appreciated indeed ! If I may add some thoughts :

- "scale" should be auto filled from the Techdraw::Page object
- "organization" should be auto filled from the "Edit->Preferences->General->Document->Authoring and License" section, as long as the "author" field
- "date" should be auto filled to datetime.now()
- "name" should be auto filled to the current document name I'd say
-  "drawing number" or "project name" : IMHO FreeCAD currently lacks a way to set proper project name. When a project is built on multiple documents (i.e. like a program build from multiple source files), we should have a way to set the project name somewhere. We could use a textfile to store such data, and place it at the project root dir. I've written a [mod](https://github.com/nm2107/FreeCADModImportGlobalVars) which uses a `FCProject` file placed at the project's root dir to store some infos about the project, and I think it could be enhanced to also contain the project name. Text file are easy to version, exchange and manipulate.
- "material" and "weight" : auto guessed from the items in the view

All inputs should be editable even if auto filled IMHO.

luzpaz, 2024-03-06:
@pavltom any interest in this issue ?

luzpaz, 2024-03-08:
Many thanks @pavltom. I'm assigning this to you in that case. Cheers!

closed 2024-03-23 by WandererFan

Roundwolf, 2024-08-29:
Hi @pavltom, it would be helpful if it would be possible to define the source for the autofill information by yourself. For example, like you refer to the "Created By" field or the "Company" field in the base properties, you could add your own property to the base properties and could refer to it in your drawing template by the label of the property. This would also allow you to all the other properties in BaseProperties. I hope that wasn't too confusing. 

Furthermore, the material should automatically fill in with the upcoming material workbench.

luzpaz, 2024-08-29:
@Roundwolf open a Feature request ?

nm2107, 2024-11-22:
Hello guys,

Thank you for the autofill feature. It works nicely when adding a new page. However the autofilled fields do not seem to be automatically updated when the page is edited (e.g. when changing the page's scale, the scale field is not automatically updated accordingly). Does this automatic autofill feature has been discussed yet ?

IMHO the autofilled fields should reflect the latest changes, i.e. :

- when adding a new TD page, update the other TD pages in the document for sheet count
- when the model is changed, or the TD is modified, update the "date" field
- when the author or organization name is changed in the app preferences, update these fields in the techdraw pages of the document
- when the page scale is changed, update the "scale" field accordingly
- when the document's name is changed, update the "title" field in the techdraw pages of this document

WDYT ? It would improve UX and avoid a manual step to sync such field (as for now we have to click the field to update it with the autofill value)