FreeCAD/FreeCAD #5906: Preferences, General, Units, user system MKS(m/kg/s/degree) displaying dimensions as millimetres and not metres

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3922

* **Reporter:** ArnoVersfeld
* **Date submitted:** 3/29/2019
* **FreeCAD version:** 0.18
* **Category:** Bug
* **Status:** feedback
* **Tags:** units

# Original report text

I installed FreeCAD and set the Preferences as shown in the summary.
When I go to Arch, Draft or Drawing workbench and draw a simple image like a square then use the dimension tool I get a display in millimetres and not metres. I closed FreeCAD.
I deleted the file in the .FreeCAD folder namely user.cfg and cookies and opened FreeCAD again and got the same results.
I uninstalled FreeCAD from the system and reinstalled it after again deleting the files in the FreeCAD folder and got the same results, namely, "display in millimetres and not metres".
When I go to the combo view of a rectangle that I've drawn the Height is 8,00 mm and the Length 12,00 mm these dimensions should be 0,80 m and 1,20 m.

# Additional information

Contents of the user.cfg file:
```
<?xml version="1.0" encoding="UTF-8" standalone="no" ?>
<FCParameters>

  <FCParamGroup Name="Root">
    <FCParamGroup Name="BaseApp">
      <FCParamGroup Name="LogLevels">
        <FCInt Name="Default" Value="2"/>
      </FCParamGroup>
      <FCParamGroup Name="Preferences">
        <FCParamGroup Name="General">
          <FCText Name="FileOpenSavePath">/home/arno</FCText>
          <FCText Name="AutoloadModule">StartWorkbench</FCText>
          <FCBool Name="ShowSplasher" Value="1"/>
          <FCBool Name="PythonWordWrap" Value="1"/>
          <FCInt Name="ToolbarIconSize" Value="24"/>
        </FCParamGroup>
        <FCParamGroup Name="Units">
          <FCInt Name="UserSchema" Value="1"/>
          <FCInt Name="Decimals" Value="2"/>
          <FCInt Name="FracInch" Value="8"/>
        </FCParamGroup>
        <FCParamGroup Name="Macro">
          <FCBool Name="LocalEnvironment" Value="1"/>
          <FCText Name="MacroPath">/home/arno/.FreeCAD/Macro</FCText>
          <FCBool Name="RecordGui" Value="1"/>
          <FCBool Name="GuiAsComment" Value="1"/>
          <FCBool Name="ScriptToPyConsole" Value="1"/>
          <FCBool Name="ScriptToFile" Value="0"/>
          <FCText Name="ScriptFile">FullScript.FCScript</FCText>
        </FCParamGroup>
        <FCParamGroup Name="Mod">
          <FCParamGroup Name="OpenSCAD"/>
          <FCParamGroup Name="Import">
            <FCParamGroup Name="hSTEP">
              <FCBool Name="ReadShapeCompoundMode" Value="1"/>
            </FCParamGroup>
          </FCParamGroup>
          <FCParamGroup Name="Start">
            <FCText Name="AutoloadModule">StartWorkbench</FCText>
            <FCUInt Name="BackgroundColor1" Value="1331197183"/>
            <FCUInt Name="BackgroundTextColor" Value="4294703103"/>
            <FCUInt Name="PageColor" Value="4294967295"/>
            <FCUInt Name="PageTextColor" Value="255"/>
            <FCUInt Name="BoxColor" Value="3722305023"/>
            <FCUInt Name="LinkColor" Value="65535"/>
            <FCUInt Name="BackgroundColor2" Value="2141107711"/>
            <FCText Name="Template"></FCText>
            <FCText Name="BackgroundImage"></FCText>
            <FCText Name="ShowCustomFolder"></FCText>
[... 373 lines cut ...]

---

### Comment by Kunda1 2019-07-05 21:08
fixed typos in title

---

### Comment by Kunda1 2019-07-05 21:09
ping ArnoVersfeld

---

### Comment by Kunda1 2019-08-29 23:26
chrisb is there an already open issue or known issue related to this ticket ?

---

### Comment by chrisb 2021-02-16 07:58
No forum discussion, not tested in 0.19, no file, only preferences shown as XML and no screenshot? Hard to validate.

---

### Comment by wmayer 2021-02-23 18:59
If you choose MKS in the unit preferences and create a standard box afterwards it shows 10mm x 10mm x 10mm. The reporter probably expects 0.01m x 0.01m x 0.01m

Vanuan, 2024-02-15:
Looks like the culprit is here: https://github.com/FreeCAD/FreeCAD/issues/7395#issuecomment-1227220095

So MKS is not meters but "meters or millimeters" depending on the value.

closed 2025-06-08 by github-actions[bot]

luzpaz, 2025-06-08:
Units getting addressed now. reopening so we can mindfully close it.