FreeCAD/FreeCAD #5915: Improved Sketcher Messages

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3935

* **Reporter:** abdullah
* **Date submitted:** 4/16/2019
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** 

# Original report text

a) Informing (solver messages):

Unconstrained sketch with 1 DoF / Fully Constrained sketch (As today)
Some constraints are partially redundant (in blue, for information)
Solved in 0.5 seconds (As today)

b) In the constraints list "mark" somehow those constraints which are partially redundant.

https://forum.freecadweb.org/viewtopic.php?p=301281#p301231

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 12/29/2021

# Discussion from Mantis ticket

### Comment by chennes 2021-12-29 05:17
I believe that the first part of this feature request is resolved now, still leaving the second part (marking the partially-redundant constraints).

luzpaz, 2022-12-03:
What's the status of this ticket?

abdullahtahiriyo, 2022-12-04:
@luzpaz 

Could you ping Chrisb here for checking? (I do not manage to get the right Chris). Thanks!

luzpaz, 2022-12-04:
CC @PLChris

abdullahtahiriyo, 2022-12-10:
Thank you Chris. I had not even realised I had opened it. Closing...

closed 2022-12-10 by abdullahtahiriyo