FreeCAD/FreeCAD #5916: DataAlongLine Create Plot Always plot Magnitude

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3937

* **Reporter:** salmat
* **Date submitted:** 4/18/2019
* **FreeCAD version:** 0.18
* **Category:** Bug
* **Status:** assigned
* **Tags:** 

# Original report text

In DataAlongLine
The "Create Plot" Always plot only Magnitude. Witch ever Vector your are chosen to plot.
I need to have only X data for example.

I have seen the bug in any versions.

I have looks the code, it seems to me it is in :
src/Mod/Fem/App/FemPostFilter.cpp line 267

When not "if(pdata->GetNumberOfComponents() == 1)"
It always does the mean square, not looking selected Vector value
(Not sure I understand everything ...)

line 267 :
```void FemPostDataAlongLineFilter::GetAxisData() {

    std::vector<double> coords;
    std::vector<double> values;

    vtkSmartPointer<vtkDataObject> data = m_probe->GetOutputDataObject(0);
    vtkDataSet* dset = vtkDataSet::SafeDownCast(data);
    vtkDataArray* pdata = dset->GetPointData()->GetArray(PlotData.getValue());
    vtkDataArray *tcoords = dset->GetPointData()->GetTCoords("Texture Coordinates");

    int component = 0;

    const Base::Vector3d& vec1 = Point1.getValue();
    const Base::Vector3d& vec2 = Point2.getValue();
    const Base::Vector3d diff = vec1 - vec2;
    double Len = diff.Length();

    for(int i=0; i<dset->GetNumberOfPoints(); ++i) {

        double value = 0;
        if(pdata->GetNumberOfComponents() == 1)
            value = pdata->GetComponent(i, component);
        else {
            for(int j=0; j<pdata->GetNumberOfComponents(); ++j)
                value += std::pow(pdata->GetComponent(i, j),2);

            value = std::sqrt(value);
        }
        values.push_back(value);
        double tcoord = tcoords->GetComponent(i, component);
        coords.push_back(tcoord*Len);
    }
    YAxisData.setValues(values);
    XAxisData.setValues(coords);
}```

# Additional information

OS: Ubuntu 18.04.2 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16093 (Git)
Build type: Release
Branch: releases/FreeCAD-0-18
Hash: 690774c0effe4fd7b8d2b5e2fb2b8c8d145e21ce
Python version: 2.7.15rc1
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: French/France (fr_FR)

# Steps to reproduce

Attached an project example, with the print screens showing the bug.

1: Open the example
2: double click on DataAlongLine in Pipeline
3: in Field, select StrainVectors
4: in Vector, select any value

In 3d Print, all seems to plot normaly along the line colors, but in AlonLine plot (curve) always shows mean square ...

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: ubuntu 18.04 LTS**
* **Platform:** x86-64
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2019-04-22 13:56
salmat thank you for the report. We ask folks to first post on the forum before opening a ticket, especially in regards to FEM because it is heavily developed. 

Also consider using the 0.19 development version when you're using FEM.

punting to bernd

---

### Comment by salmat 2019-04-22 20:42
How to install 0.19 ?
I have looked : https://www.freecadweb.org/wiki/Install_on_Unix/id
Installed freecad-daily, and just tried to update now ... nothing new ?

---

### Comment by Kunda1 2019-04-23 11:21
salmat Try the appimages. 
I guess I need to update the documentation to help folks understand this better. Here is the link to the latest dev release: https://github.com/FreeCAD/FreeCAD/releases/tag/0.19_pre
You can use the appimage update tool to stay up to date with the latest dev release. That is what the FEM dev recommends to have the FEM workbench working correctly.

luzpaz, 2022-05-04:
@FEA-eng do you mind trying to reproduce on v0.20dev ?

FEA-eng, 2022-05-04:
> @FEA-eng do you mind trying to reproduce on v0.20dev ?

I checked this and it still happens (with displacement components) in 0.20.

> OS: Windows 10 (10.0)
> Word size of FreeCAD: 64-bit
> Version: 0.20.28774 (Git)
> Build type: Release
> Python 3.8.13, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
> Locale: Polish/Poland (pl_PL)

luzpaz, 2022-09-07:
Anyone willing to fix this?

luzpaz, 2022-09-07:
Discussion thread at https://forum.freecadweb.org/viewtopic.php?f=18&t=71709

luzpaz, 2023-01-08:
Here is the updated link to the code snippet mentioned above: https://github.com/FreeCAD/FreeCAD/blob/664b2560adcd9c9ab012e4f8c0fb92143c202fe9/src/Mod/Fem/App/FemPostFilter.cpp#L314

luzpaz, 2023-01-08:
@berndhahnebach do you have  moment to look in to this ?

luzpaz, 2023-10-11:
@FEA-eng can you please test #10992 ?

commit 56422b17 (2023-10-12): Fem: Plot vector components in FemPostDataAlongLineFilter - fixes #5916
closed 2023-10-12 by wwmayer