FreeCAD/FreeCAD #5917: Polar Pattern "transformed shapes do not intersect support" (upstream OCC bug)

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3938

* **Reporter:** Sembazuru
* **Date submitted:** 4/19/2019
* **FreeCAD version:** 0.18
* **Category:** Bug
* **Status:** new
* **Tags:** #ready-for-migration, OCC 7.4, OCC Bug, OCC NotOnRoadmap

# Original report text

As discussed on the forum [here](https://forum.freecadweb.org/viewtopic.php?style=4&f=3&t=35699), there is a bug when using polar pattern if features are lined up just wrong. I tried using polar pattern of Grooves on a Revolved solid and get "transformed shapes do not intersect support". Selecting different numbers of occurrences get different array elements working, but never all of them.

As mentioned by forum member kisolre,
[quote=kisolre]We again hit some OCC coplanar/colinear/co_something issue. Moving the center of the arc some distance away from the edge of the base coin fixes the issue (see the 0.1mm dimension).[/quote]

kisolre provides a good screenshot of the "fix" to get around the bug in the forum thread.

# Additional information

This ticket is being tracked via [OCC Bugs in the Bugtracker](https://forum.freecadweb.org/viewtopic.php?f=10&t=20264) forum thread

# Steps to reproduce

* Open the attached file.
* Edit the ScallopPolarPattern and see the preview. (Also see my screenshot on my forum OP.)
* Change the number of Occurrences and see the preview.
* To get around this:

* Open the ScallopSketch to edit.
* On the horizontal construction line remove the horizontal constraint.
* Add a vertical distance constraint between the two endpoints of the horizontal construction line.
* Set that vertical distance constraint to a non-zero number.
* Close the sketch and see on the object that all the Grooves are now properly patterned around the disk.

# FreeCAD Info

```
OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16110 (Git)
Build type: Release
Branch: (HEAD detached at upstream/releases/FreeCAD-0-18)
Hash: f7dccfaa909e5b9da26bf50c4a22ccca9bb10c40
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **OS: 10 Pro 1809**
* **Platform:** Windows 64-bit
* **Updated:** 1/24/2022

# Discussion from Mantis ticket

### Comment by Kunda1 2019-04-24 11:38
Putting this on the 0.19 roadmap so that it gets some attention. 

---

### Comment by Kunda1 2020-12-22 12:42
Re-uploaded broken attachment (used file from forum thread)

---

### Comment by chennes 2021-08-25 03:43
This is still failing, as of:

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.25541 (Git)
Build type: Debug
Branch: master
Hash: ca043dcb6d20d70b8e24d75e810683214eecc5d4
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

---

### Comment by Kunda1 2022-01-15 13:52
This needs an upstream ticket

luzpaz, 2022-05-26:
@charlemagnelasse this could also use an upstream OCC ticket, if you don't mind? :wink:

CharlemagneLasse, 2022-05-27:
I've tried to reproduce it and I can reproduce it with the freecad polar pattern (2) but not with two cut bodies (one is the mirror of the other or a polar pattern of the cut body). Still trying to figure out why this is the case

But maybe I am already diverting myself with details. I have now just used a base body and a cut "body". And used as reproducer:

````
# get most recent OCCT
$ cd /tmp/
$ git clone https://git.dev.opencascade.org/repos/occt.git OCCT
$ git -C OCCT describe
V7_1_0_beta-2420-g84d39c06fa

# get build dependencies based on opencascade package in the Debian package source (deb-src) repository. will most likely work differently on whatever Distro you are using
$ sudo apt build-dep opencascade

# build everything (I only have 4 cores - you might change it if you are not working on a system with a 10 year old CPU)
$ cd OCCT
$ mkdir build
$ cd build
$ cmake .. -DCMAKE_BUILD_TYPE=Debug -DCMAKE_CXX_COMPILER_LAUNCHER=ccache -DCMAKE_VERBOSE_MAKEFILE=true && make -j4

# prepare reproducer script
$ cd /tmp/OCCT/
$ cat > test.script << "EOF"
# init modules
pload ALL

# load
restore /tmp/ScallopedEdgeCoin/ScallopedEdgeCoin_Simple-base-body.brp base
restore /tmp/ScallopedEdgeCoin/ScallopedEdgeCoin_Simple-cut-body.brp cut

bcut result base cut
save result result.brp 

# optional:
# vinit
# vdisplay result
# vfit
EOF

# start DRAWEXE shell
$ LD_LIBRARY_PATH=/tmp/OCCT/build/lin64/gcc/libd/ CASROOT=/tmp/OCCT gdb -q --ex run --args ./build/lin64/gcc/bind/DRAWEXE -f test.script

# just if somebody wondered - it didn't crash :)
````
[ScallopedEdgeCoin.zip](https://github.com/FreeCAD/FreeCAD/files/8785103/ScallopedEdgeCoin.zip)

And as it turns out, this looks perfectly fine with this version of occt. I can check if I can still reproduce it this way with OCCT 7.5.3

CharlemagneLasse, 2022-05-27:
I was able to reproduce this with 7.5.3. I've started bisecting this problem via:

````
git bisect start --term-new fixed --term-old broken
git bisect broken V7_5_3
git bisect fixed 84d39c06fa79d4227c1bc415e36e3924f7ff64c6
````

This is my bisect log:

````
git bisect start '--term-new' 'fixed' '--term-old' 'broken'
# broken: [fecb042498514186bd37fa621cdcf09eb61899a3] Update version up to 7.5.3
git bisect broken fecb042498514186bd37fa621cdcf09eb61899a3
# fixed: [84d39c06fa79d4227c1bc415e36e3924f7ff64c6] 0032973: Modeling Algorithms - Regression in BRepExtrema_DistShapeShape compared with 7.5
git bisect fixed 84d39c06fa79d4227c1bc415e36e3924f7ff64c6
# broken: [628c0211d53c7fe1036a85e7a7b2b067c9c50f7a] Increment OCCT version up to 7.5.0
git bisect broken 628c0211d53c7fe1036a85e7a7b2b067c9c50f7a
# broken: [0f04bd99d07f417abbecf17d54c4a9754aa4f3ae] 0032475: Configuration, qmake - drop "-z defs" linker flag on macOS
git bisect broken 0f04bd99d07f417abbecf17d54c4a9754aa4f3ae
# broken: [d62df471b262ab4163625ab99ba653e1c0b15f17] 0032677: Coding Rules, Geom_BezierSurface - eliminate GCC warning -Wmaybe-uninitialized -fsanitize=undefined
git bisect broken d62df471b262ab4163625ab99ba653e1c0b15f17
# broken: [b66cf6dae5b39ae74d8e1bc7f99bc9a235084560] 0032832: Coding - get rid of unused headers [FairCurve to GeomAPI]
git bisect broken b66cf6dae5b39ae74d8e1bc7f99bc9a235084560
# fixed: [ee6581adbea1b030be8c10ea3763341efdc94a69] 0032874: IsParallel() method of Extrema_ExtCC does not give same results in OCC 7.6.0 vs OCC 7.5.0
git bisect fixed ee6581adbea1b030be8c10ea3763341efdc94a69
# fixed: [7573a45deb1a2d84632e7bbd373222288de682eb] 0032870: Tests - bugs/caf/bug31918_1 is unstable
git bisect fixed 7573a45deb1a2d84632e7bbd373222288de682eb
# fixed: [b9184c271493f25ae444304a62f35d9c236c6f12] 0032834: Configuration, CMake - DESTDIR ignored for OpenCASCADECompileDefinitionsAndFlags-${CMAKE_INSTALL_CONFIG_NAME_LOWER}.cmake
git bisect fixed b9184c271493f25ae444304a62f35d9c236c6f12
# broken: [600ee856316c1a9eb340779b32de511b74465896] 0032814: Modeling algorithms - Unifysamedom produces invalid result
git bisect broken 600ee856316c1a9eb340779b32de511b74465896
# fixed: [b91e601442d983e76c63bb54039193e9f35e94d8] 0032785: Application Framework - CDF_Store::SetFolder incorrectly processes relative paths on Linux
git bisect fixed b91e601442d983e76c63bb54039193e9f35e94d8
# fixed: [cfde7eebb37b82900f29fd5da99a008bac8d5aa3] 0024418: Modeling Algorithms - Wrong section curves when intersecting analytical surfaces
git bisect fixed cfde7eebb37b82900f29fd5da99a008bac8d5aa3
# first fixed commit: [cfde7eebb37b82900f29fd5da99a008bac8d5aa3] 0024418: Modeling Algorithms - Wrong section curves when intersecting analytical surfaces
````

So the problem was fixed by 

````
cfde7eebb37b82900f29fd5da99a008bac8d5aa3 is the first fixed commit
commit cfde7eebb37b82900f29fd5da99a008bac8d5aa3
Author: knosulko <knosulko@opencascade.com>
Date:   Mon Nov 15 17:02:03 2021 +0300

    0024418: Modeling Algorithms - Wrong section curves when intersecting analytical surfaces
    
    IntPatch_ALineToWLine.cxx - setting minimal number of points in a WLine depending on step size
    IntPatch_WLineTool.cxx - fix WLines connecting check condition
    test cases has been changed according to new behavior

 src/IntPatch/IntPatch_ALineToWLine.cxx | 64 ++++++++++++++++++++++++++--
 src/IntPatch/IntPatch_WLineTool.cxx    | 76 ++++++++++++++++++++++++++++++----
 tests/bugs/modalg_1/bug10606_1         |  2 +-
 tests/bugs/modalg_1/bug10606_2         |  2 +-
 tests/bugs/modalg_1/bug10606_3         |  2 +-
 tests/bugs/modalg_1/bug10606_4         |  2 +-
 tests/bugs/modalg_1/bug10606_5         |  2 +-
 tests/bugs/modalg_1/bug10842_5         |  2 +-
 tests/bugs/modalg_2/bug20964_1         |  2 +-
 tests/bugs/modalg_2/bug20964_2         |  2 +-
 tests/bugs/modalg_2/bug20964_3         |  2 +-
 tests/bugs/modalg_2/bug20964_4         |  2 +-
 tests/bugs/modalg_2/bug20964_5         |  2 +-
 tests/bugs/modalg_5/bug25715_3         |  4 +-
 tests/bugs/modalg_6/bug28626_3         |  2 +-
 tests/lowalgos/intss/bug24418_1        |  2 -
 16 files changed, 143 insertions(+), 27 deletions(-)
````

The upstream ticket can be found under https://tracker.dev.opencascade.org/view.php?id=24418
There is only a commit for this in the 7.7.0 development branch - but there is no release with this fix

AjinkyaDahale, 2022-06-23:
> @CharlemagneLasse Thanks for sleuthing this! I've added it to the OCC7.7 category in https://forum.freecadweb.org/viewtopic.php?f=10&t=20264

you didn't reference the upstream ticket for this there.

luzpaz, 2022-06-23:
@AjinkyaDahale oops :blush: OK, it's now added. Thanks!

luzpaz, 2022-11-07:
Good news: Upstream has fixed this issue in OCC v7.7.0 
See https://tracker.dev.opencascade.org/view.php?id=24418  
Ref: https://forum.freecadweb.org/viewtopic.php?f=10&t=20264

closed 2022-11-07 by luzpaz