FreeCAD/FreeCAD #5922: PartDesign Loft test failures on i386

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3949

* **Reporter:** locutusofborg
* **Date submitted:** 4/29/2019
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** assigned
* **Tags:** 

# Original report text

As you can see in the following test log, the newly created

testSimpleAdditiveLoftCase (PartDesignTests.TestLoft.TestLoft) ... FAIL
testSimpleSubtractiveLoftCase (PartDesignTests.TestLoft.TestLoft) ... FAIL

fails on i386... 

Can you please have a look? thanks!

https://objectstorage.prodstack4-5.canonical.com/v1/AUTH_77e2ada1e7a84929a74ba3b87153c0ac/autopkgtest-eoan/eoan/i386/f/freecad/20190429_060906_d8c75@/log.gz

This happens since its implementation, in commit e1b78ab91f86f259dc268cb742b4dc42dae7fcea

We started having that failure in 0.17, so it probably never worked on i386 platform

# Steps to reproduce

Run tests on a 32bit

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Ubuntu 18.04 and later 32 bit**
* **Platform:** Linux
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2019-04-29 12:09
locutusofborg thank you for the ticket. Please follow the guidelines next time (yellow banner top of page). Please open a forum thread to discuss with the devs and testers. Thank you.

---

### Comment by locutusofborg 2019-04-29 12:54
Kunda1 thanks! I also sent an email to the original developer of the test (found via git log) and pointed him to this bug!

---

### Comment by Kunda1 2019-10-17 22:51
locutusofborg who is the original author ? The title of this ticket is too generic

---

### Comment by locutusofborg 2020-01-13 12:07
hello Kunda1, according to git log

From e1b78ab91f86f259dc268cb742b4dc42dae7fcea Mon Sep 17 00:00:00 2001
From: Kurt Kremitzki <kkremitzki@gmail.com>
Date: Wed, 28 Jun 2017 11:31:34 -0500
Subject: [PATCH] Add basic PartDesign::Loft test cases

https://github.com/FreeCAD/FreeCAD/commit/e1b78ab91f86f259dc268cb742b4dc42dae7fcea.patch

adrianinsaval, 2022-03-10:
AFAIK this test is not failing nowadays, issue should be closed IMO.

adrianinsaval, 2022-03-10:
Although looking at it better I don't think we're testing on i386 nowadays... does FreeCAD officially support it?

Roy-043, 2023-07-16:
Since FreeCAD 0.19 we no longer officially support 32bit Windows. Closing this as stale.

closed 2023-07-16 by Roy-043