FreeCAD/FreeCAD #5925: Boolean Cut not working correctly when using arrays (polar in this case)

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3958

* **Reporter:** jr222
* **Date submitted:** 5/3/2019
* **FreeCAD version:** 0.18
* **Category:** Bug
* **Status:** new
* **Tags:** array, boolean

# Original report text

When doing a boolean difference between an object and an array of objects, some of the objects within the array seem to not cut correctly, while others do.

I posted a [forum question in the Freecad forum](https://forum.freecadweb.org/viewtopic.php?f=3&t=36090), and the problem was reproduced by most users that tested it.  Some supplied workarounds, but these are not possible for more complex projects.

This does look like a bug, as it is reproducible.  It seems that the more complex the compound objects, the easier it is to reproduce.

The example is the inner portion of a ball bearing model for 3D printing.

# Additional information

The model file and a screenshot of the problem are attached.

The reproducibility of the initial simpler problem posted in the forum is "sometimes", this more complex model should result in the bug being reproduced "always".

# Steps to reproduce

This example has the complexity required to reproduce the bug (most operations in the part workbench - parameters mentioned are changed from the default):
1. Create a torus (r1=36mm, r2=4mm)
2. Create a half sphere (rad=5mm, angle3=180, placement-angle=90, position.X=40mm)
3. Use Draft Array to make a polar array from the half spheres (type=polar, number.polar=16)
4. Fuse the torus and half-sphere-array (using boolean union)
5. Create a smaller full sphere (rad=4.5mm, position.X=40mm)
6. 3. Use Draft Array to make a polar array from the full smaller spheres (type=polar, number.polar=16)
7. Cut the smaller full sphere array from the Fusion object from step 4 using Boolean difference
8. Create a cylinder (rad=37.5mm, position.Z=-5mm)
9. Cut the Cylinder from the "Cut" object from step 7 using Boolean difference
Note that some of the smaller balls did not cutout the inside of the compound object.
(in the example I copied, I changed the NumberX and NumberY values of the arrays from 2 to 1, but that I believe should not have changed the result)

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **OS: Windows 10 10.0.17134.706**
* **Platform:** Windows
* **Updated:** 8/25/2021

# Discussion from Mantis ticket

### Comment by jr222 2019-05-03 03:52
I re-installed Freecad 0.17 and was able to reproduce the problem in that version.  i suspect this problem has been there for a while.
I also selected major because this looks to me like a bug with substance, however, please feel free to change this, as I dont have the reference to know a major from a minor.

---

### Comment by sariug 2019-10-15 12:45
There seem to be no problem with mine. 

OS: Ubuntu 18.04.3 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16093 (Git)
Build type: Release
Branch: releases/FreeCAD-0-18
Hash: 690774c0effe4fd7b8d2b5e2fb2b8c8d145e21ce
Python version: 2.7.15rc1
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

---

### Comment by openBrain 2019-10-15 17:50
Issue still present in 0.19 (also notice that new Draft/Array implementation breaks the example file)
OS: Ubuntu 18.04.3 LTS (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.18502 (Git)
Build type: Unknown
Branch: master
Hash: ca30dfeb27871352f0adc4dae067fd5ec7b06670
Python version: 3.6.8
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: French/France (fr_FR)

---

### Comment by chennes 2021-08-25 17:36
Possibly related to https://tracker.freecadweb.org/view.php?id=3938

Jolbas, 2023-03-30:
I can't reproduce.

OS: macOS 12.6
Word size of FreeCAD: 64-bit
Version: 0.21.0.32630 (Git)
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)

luzpaz, 2023-03-30:
Is it related to #5917 ?

Jolbas, 2023-03-31:
I would say probably. Here's my test file from following the reproducing steps.
[issue5925.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/11118474/issue5925.FCStd.zip)

Roy-043, 2023-07-16:
I can't reproduce either.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.21.0.33492 (Git)
Build type: Release
Branch: master
Hash: 6e235a4809e14674aa2cd23e68630b68998a53d4
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

closed 2023-07-16 by Roy-043