FreeCAD/FreeCAD #5927: DogLeg fail

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3961

* **Reporter:** kisolre
* **Date submitted:** 5/6/2019
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** confirmed
* **Tags:** 

# Original report text

Following the request in the Report View here is a situation where DogLeg fails but LevenbergMarquardt succeeds

# Additional information

```OS: Windows 8.1
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.16624 (Git)
Build type: Release
Branch: master
Hash: 222ae7305fdf1097e4ef3d050f69dff47dbd8786
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: Bulgarian/Bulgaria (bg_BG)
```

# Steps to reproduce

Repeat the steps in the video in the following post with the file m_constrained_2.fsctd from the same post: [Post](https://forum.freecadweb.org/viewtopic.php?p=306134#p306134)
1.Open sketch01
2.Change dimmension 10mm to 24mm
3.Change dimmension 6mm to 20mm
4.Exit sketch
In report view I get:
```EigenSparseQR, Threads: 1, Vectorization: On, Pivot Threshold: 1e-13, Params: 12, Constr: 12, Rank: 12
Sketcher::setUpSketch()-T:0.081
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
Sketcher::Solve()-DogLeg-T:0
EigenSparseQR, Threads: 1, Vectorization: On, Pivot Threshold: 1e-13, Params: 12, Constr: 12, Rank: 12
Sketcher::setUpSketch()-T:0.019
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL, Iteration: 0, fx_inf(tolf): 13.9, g_inf(tolg): 13.9, delta(f(tolx)): 0.3, err(divergingLim): 96.605
DL, Iteration: 1, fx_inf(tolf): 13.6, g_inf(tolg): 13.6, delta(f(tolx)): 0.9, err(divergingLim): 92.48
DL, Iteration: 2, fx_inf(tolf): 12.7, g_inf(tolg): 12.7, delta(f(tolx)): 2.7, err(divergingLim): 80.645
DL, Iteration: 3, fx_inf(tolf): 10, g_inf(tolg): 10, delta(f(tolx)): 8.1, err(divergingLim): 50
DL, Iteration: 4, fx_inf(tolf): 1.9, g_inf(tolg): 1.9, delta(f(tolx)): 24.3, err(divergingLim): 1.805
DL, Iteration: 5, fx_inf(tolf): 0, g_inf(tolg): 0, delta(f(tolx)): 24.3, err(divergingLim): 0
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
Sketcher::Solve()-DogLeg-T:0.011
EigenSparseQR, Threads: 1, Vectorization: On, Pivot Threshold: 1e-13, Params: 12, Constr: 12, Rank: 12
Sketcher::setUpSketch()-T:0
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL, Iteration: 0, fx_inf(tolf): 13.9, g_inf(tolg): 13.9, delta(f(tolx)): 0.3, err(divergingLim): 96.605
DL, Iteration: 1, fx_inf(tolf): 13.6, g_inf(tolg): 13.6, delta(f(tolx)): 0.9, err(divergingLim): 92.48
DL, Iteration: 2, fx_inf(tolf): 12.7, g_inf(tolg): 12.7, delta(f(tolx)): 2.7, err(divergingLim): 80.645
DL, Iteration: 3, fx_inf(tolf): 10, g_inf(tolg): 10, delta(f(tolx)): 8.1, err(divergingLim): 50
DL, Iteration: 4, fx_inf(tolf): 1.9, g_inf(tolg): 1.9, delta(f(tolx)): 24.3, err(divergingLim): 1.805
DL, Iteration: 5, fx_inf(tolf): 0, g_inf(tolg): 0, delta(f(tolx)): 24.3, err(divergingLim): 0
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
Sketcher::Solve()-DogLeg-T:0
EigenSparseQR, Threads: 1, Vectorization: On, Pivot Threshold: 1e-13, Params: 12, Constr: 12, Rank: 12
Sketcher::setUpSketch()-T:0
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
DL: tolg: 1e-80, tolx: 1e-80, tolf: 1e-10, convergence: 1e-10, dogLegGaussStep: LeastNormLdlt, xsize: 1, csize: 1, maxIter: 100
DL: stopcode: 1, Success
Sketcher::Solve()-DogLeg-T:0
Sketcher::setUpSketch()-T:0.02
Sketcher::Solve()-DogLeg-T:0.011
Sketcher::setUpSketch()-T:0.002
Sketcher::Solve()-DogLeg- Failed!! Falling back...
Important: the LevenbergMarquardt solver succeeded where the DogLeg solver had failed.
If you see this message please report a way of reproducing this result at
http://www.freecadweb.org/tracker/main_page.php
Sketcher::Solve()-LevenbergMarquardt-T:0.007
Hide main window
Show main window
Hide main window
Show main window
```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 2/6/2021

adrianinsaval, 2023-07-03:
@AjinkyaDahale do you consider this issue relevant or should it be closed?

AjinkyaDahale, 2023-07-13:
@adrianinsaval sorry your tag somehow missed my attention.

Just one solver failing and other succeeding doesn't seem to be a problem (even if the error reports should be shortened as discussed in another issue I can't seem to find). However, this is evidently seen as a problem at the `Sketch.cpp` level. The code (at `src/Mod/Sketcher/App/Sketch.cpp:4329`) was last edited more than a decade ago by a user named "logari" in 67018ac2112f7ea9291f579709900acddbb826a5.

I'll leave the decision to close up to @abdullahtahiriyo. My opinion is that this shouldn't be much of an issue for the users, though the test cases may have some academic value. If we close this issue perhaps we should consider removing this line as well. The message could be something along the lines of "something interesting happened and we would like to know more about it" as opposed to the current appearance of "something wrong happened so please report it".

abdullahtahiriyo, 2023-11-28:
It is never a problem for the user if he gets a solution he wants. The issue somehow, is that DogLeg, being on paper a superior solution, should not fail, warranting to look into it. However, it is not trivial to debug DogLeg, and we do not even have some nice debugging code.

Perhaps it is time to move on. Close this ticket and remove the request for reporting from the code.

AjinkyaDahale, 2023-11-28:
> Perhaps it is time to move on. Close this ticket and remove the request for reporting from the code.

I agree with this. Though we may want to start using those new C++ tests.

abdullahtahiriyo, 2023-11-29:
I am not so sure these qualify as suitable tests in the sense that the input is not direct to DogLeg. Not even direct to Sketch Interface. But Sketcher. Of course, fine if the input is "converted to the DogLeg input", but then the test is not trivial to understand, as the DogLeg input does not correspond (directly) to a physical meaning of the sketch.

When this in code tests started to trend, I thought about it and reached the conclusion that the information we have is better suited for integration tests (so the tests run by the CI on every PR). Of course, it has the disadvantage that  the feedback loop is not as fast. But I may be wrong. I did not finish my book yet...

Of course, if we come with a good way, why not?

AjinkyaDahale, 2023-11-29:
> I am not so sure these qualify as suitable tests in the sense that the input is not direct to DogLeg. Not even direct to Sketch Interface. But Sketcher.

The C++ tests run by themselves without starting freecad. I believe we should be able to directly import planeGCS and play with it.

> Of course, fine if the input is "converted to the DogLeg input", but then the test is not trivial to understand, as the DogLeg input does not correspond (directly) to a physical meaning of the sketch.

> When this in code tests started to trend, I thought about it and reached the conclusion that the information we have is better suited for integration tests (so the tests run by the CI on every PR). [...]

What book?

abdullahtahiriyo, 2023-12-04:
There was a recommended book in the forum regarding testing.

AjinkyaDahale, 2023-12-06:
> There was a recommended book in the forum regarding testing.

Tried searching but couldn't find (just 10-15 minutes though). Do you happen to recall where this was?

abdullahtahiriyo, 2023-12-06:
Nope, but the book is titled "Working Effectively with Legacy Code" by Michael C. Feathers.

AjinkyaDahale, 2023-12-06:
OK name was the more important part anyway.

FEA-eng, 2025-09-14:
@AjinkyaDahale Is it still relevant?

luzpaz, 2026-09-22:
Another reason for: 
* #5752