FreeCAD/FreeCAD #5928: Only alpha comma key can be used as decimal separator

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3963

* **Reporter:** openBrain
* **Date submitted:** 5/8/2019
* **FreeCAD version:** 0.18
* **Category:** Bug
* **Status:** resolved
* **Tags:** 0.18, input, keyboard, units

# Original report text

On some values in the Property editor panel, only the alpha comma key is accepted as decimal separator.
This is suboptimal (mainly compared to num dot key) when entering numbers
```OS: Linux Mint 19.1
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16093 (Git)
Build type: Release
Branch: releases/FreeCAD-0-18
Hash: 690774c0effe4fd7b8d2b5e2fb2b8c8d145e21ce
Python version: 2.7.15rc1
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: French/France (fr_FR)```

# Additional information

[Linked forum thread](https://forum.freecadweb.org/viewtopic.php?f=3&t=35927)

AFAIK, no FC forumer natively using dot sign as decimal separator (eg en_US) has confirmed or not the bug. Only users using comma as decimal separator did.
Proposed fix in the forum discussion : * Numpad dot key is always accepted as decimal separator
* User options to enforce FC to specifically accept alpha dot and/or alpha comma as decimal separator

# Steps to reproduce

Edit a value directly in the property editing panel.
Some will indifferently accept comma or dot (alpha comma, alpha dot, num dot) - eg Placement Angle - and some will enforce user to use alpha comma - eg Placement x Axis -.

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **Updated:** 1/3/2022
* **Fixed in version:** 0.2

# Discussion from Mantis ticket

### Comment by openBrain 2020-03-24 12:09
https://github.com/FreeCAD/FreeCAD/pull/3256

---

### Comment by openBrain 2020-03-24 12:40
For specific case of TechDraw in the forum discussion, UI files have to be updated to use 'Gui::DoubleSpinBox' instead of 'QDoubleSpinBox'

---

### Comment by uwestoehr 2020-03-24 13:25
Duplicate of #3795

---

### Comment by uwestoehr 2020-03-24 13:26
Duplicate of #3795

---

### Comment by openBrain 2020-03-24 13:57
Not duplicate. Related to.

---

### Comment by chennes 2021-12-29 05:15
I know there has been some work in this direction recently: where does this bug report stand openBrain and uwestoehr?

---

### Comment by openBrain 2022-01-03 08:51
chennes : to me the problem as described in this ticket is indeed solved. I had no complaints about the fix till now, so I guess we can close.

---

### Comment by uwestoehr 2022-01-03 11:23
Fixed in FC 0.20dev but cannot be backported to FC 0.19 -> marking the bug fixed but keep it open to be transferred to GitHub as reference for 0.19 users.

0penBrain, 2022-02-07:
@donovaly What should be done according your last comment ?

donovaly, 2022-02-07:
Leaving it open as fixed for FC 0.20.

0penBrain, 2022-02-07:
> Leaving it open as fixed for FC 0.20.

Well, OK. Don't really get the sense of it, but as you want. ;)

donovaly, 2022-02-07:
The idea is that users will find it in the bug list, see that it is already reported and fixed, they just nee to wait for the next major release.

donovaly, 2022-02-08:
As an average user, for example for me this is the case with LibreOffice. I see a bug, then I check if there is an open bug. If not, I report it. It always annoyed me when I report a bug and then I get as reply, that it is already fixed, just not in the latest release.
Then I think to myself: "please think about users. As user I take the latest stable release and f you want me to report bugs you should not disappoint me to do senseless work in reporting issues that are already known."

adrianinsaval, 2022-02-08:
IMO the correct approach is to search for open OR closed reports but it doesn't hurt to leave it open either, specially if we use the tag mentioned it is fixed, then as part of the release process of the next version all issues with this tag should be closed. The rest of the team managing issues has to be aware and agree to follow this approach though, otherwise it will be even more confusing because some bugs fixed in master are closed but others are kept open.

donovaly, 2022-06-24:
FC 0.20 was released -> fixed

closed 2022-06-24 by donovaly