FreeCAD/FreeCAD #5931: Solver falsely detects redundant constraints in specific conditions

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3976

* **Reporter:** openBrain
* **Date submitted:** 5/15/2019
* **FreeCAD version:** 0.18
* **Category:** Bug
* **Status:** confirmed
* **Tags:** 

# Original report text

In some specific conditions, solver falsely detects redundant constraints.
It happens when 2 lines/curves are tangent by design but not by constraint.

# Additional information

[Forum thread](https://forum.freecadweb.org/viewtopic.php?f=3&t=36393&start=10#p308408)
---
[Other example file that produces the same issue by chrisb](https://forum.freecadweb.org/viewtopic.php?f=3&t=36393&start=20#p308457)
---
OS: Ubuntu 18.04.2 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.1.
Build type: Release
Python version: 2.7.15rc1
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: French/France (fr_FR)

# Steps to reproduce

With minimal attached file :
*  Remove 2.5 mm constraint and try to add a radius constraint to the arc => The solver falsely detects a redundant constraint
*  Reset sketch to initial state, change 4 mm constraint to 3 mm then repeat same as previous step => The solver correctly find a fully constrained sketch

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by DeepSOIC 2020-03-29 20:05
This is the same kind of problem as with coincident+tangent combo. You have enforced top line tangent to arc by means of other constraints. The point connecting the circle and the line must be computed by finding an intersection between the two, which is an ill-posed problem for the solver when they are tangent. 

The false detection of redundancy comes from vanishing of jacobi matrix row when the sketch is precisely solved. This is also exactly what a redundancy looks like. If the solution is slightly inaccurate (e.g. if you move the point away a bit before applying radius constraint), there won't be a false redundancy message.

I'm afraid this is unfixable. Having at least some warning (i.e. a false redundancy warning) may be better than nothing, actually.

---

### Comment by openBrain 2020-03-29 21:53
DeepSOIC : I'm not sure this is exactly same case as coincident+tangent. In the reproducing procedure, a DOF is released by removing the "2.5 mm" length. This is super weird as if you follow the procedure and try to validate an arc radius (any value), it will refuse it with "sketch contains conflicting constraints".
Just change the 2.5 mm length to any other before removing it, and then any value of arc radius is correctly accepted (leading to fully constrained sketch). Even a 2.5 mm radius is correct.

0penBrain, 2023-04-26:
Just checked (to be sure) and still present with :
```
OS: Ubuntu 22.04.2 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33021 (Git)
Build type: Debug
Branch: master
Hash: c77b2f33c29b9f0f24ddadae4d43900fa720758b
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: English/United States (en_US)
```

maxwxyz, 2024-04-13:
maybe related:
- https://github.com/FreeCAD/FreeCAD/issues/5870

FEA-eng, 2026-10-19:
Still relevant

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261013 (Git shallow)
Build date: 2026/10/13 19:55:30
Build type: Release
Branch: (HEAD detached at 50daee777)
Hash: 50daee777321b5085accd1f57b57f1a7c1781cae
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 617.14/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```