FreeCAD/FreeCAD #5932: Fillet feature fails on solid obtained by revolution

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3977

* **Reporter:** openBrain
* **Date submitted:** 5/15/2019
* **FreeCAD version:** 0.18
* **Category:** Bug
* **Status:** confirmed
* **Tags:** fillet

# Original report text

Fillet function fails on particular part obtained by revolution. It correctly works on same part obtained by other function(s).

# Additional information

[Forum thread](https://forum.freecadweb.org/viewtopic.php?f=3&t=35972#p304298)
---
OS: Ubuntu 18.04.2 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.1.
Build type: Release
Python version: 2.7.15rc1
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: French/France (fr_FR)

# Steps to reproduce

Attached file demonstrates the bug with one part where filleting fails (obtained by Revolution) and one where it succeeds (obtained by Pad).
More precisely, on the working model, the 4 inner edges of the "crest" can be filleted, while only 2 are in the faulty design. Trying to fillet the other 2 will fail or even freeze FC.

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **Updated:** 2/18/2021

grubertechnic, 2023-07-29:
Still an issue.

```
OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33658 (Git)
Build type: Release
Branch: master
Hash: ba5f7e6d75dee09ec7076ad24895b04e56f21b63
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
```

Attached the file, because in old tracker it was not downloadable. Luckly openBrain have named in the forum the original file, which is attached.

[5932.zip](https://github.com/FreeCAD/FreeCAD/files/12207625/5932.zip)

FEA-eng, 2025-10-23:
Still relevant. There's even this effect for one of the edges:

[image]

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/21 12:07:13
Build type: Release
Branch: main
Hash: 2096d0bd8603b429fdc0c746c34b3d73c3289fe1
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```