FreeCAD/FreeCAD #5933: Add Expression support to certain fields

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3981

* **Reporter:** chrisb
* **Date submitted:** 5/19/2019
* **FreeCAD version:** 0.18
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

Some fields do not support expressions, see the [forum discussion](https://forum.freecadweb.org/viewtopic.php?f=3&t=33817&p=283085#p309255).
It would be natural to support them too. 

Added: The ratio in snells law constraint should be added to the list, see https://forum.freecadweb.org/viewtopic.php?p=331817#p331817.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by uwestoehr 2019-05-19 15:55
The missing expressions lead actually to bugs, take for example the case I described here: https://forum.freecadweb.org/viewtopic.php?f=3&t=36458#p308966

---

### Comment by uwestoehr 2020-03-21 04:17
The PartDesign primitives dialog edits get expressions by this PR: https://github.com/FreeCAD/FreeCAD/pull/3225

maxwxyz, 2024-03-10:
Is this still relevant?

pierreporte, 2024-04-07:
@maxwxyz Yes it would be nice to be able to put expressions everywhere.

FlachyJoe, 2024-04-07:
The forum list of missing had to be updated.
All Part feature support now Expression in the property list but not in the task panel.

luzpaz, 2025-01-25:
What's the status of this ? Is there a roadmap ?