FreeCAD/FreeCAD #5934: Security Audit of 0.19

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3983

* **Reporter:** Kunda1
* **Date submitted:** 5/19/2019
* **FreeCAD version:** 
* **Category:** General
* **Status:** new
* **Tags:** #ready-for-migration

# Original report text

Using flawfinder 2.0.8 I've generated the attached report for consideration.

This issue has been set to 'private'. 

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** General
* **Updated:** 2/3/2022

adrianinsaval, 2022-02-11:
@luzpaz what to do with this one?

luzpaz, 2022-02-11:
Keep it open for when we announce a feature-freeze and I'll run another audit  for v0.20dev

donovaly, 2022-07-23:
FreeCAD 0.20 is out. if there should be an audit, just do this and make a new PR or open a forum thread in the development section.

closed 2022-07-23 by donovaly