FreeCAD/FreeCAD #5935: Wrong simulation after setting a start point

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3989

* **Reporter:** Andrea3
* **Date submitted:** 5/28/2019
* **FreeCAD version:** 0.18
* **Category:** Bug
* **Status:** confirmed
* **Tags:** 

# Original report text

Simulation shows that the object is cut completely but seems not possible due to the tool was not commanded under a specific Z value.

# Additional information

Forum link:
https://forum.freecadweb.org/viewtopic.php?f=3&t=36523

FreeCAD version:

OS: Windows 7
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16110 (Git)
Build type: Release
Branch: (HEAD detached at upstream/releases/FreeCAD-0-18)
Hash: f7dccfaa909e5b9da26bf50c4a22ccca9bb10c40
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0

# Steps to reproduce

Draw a simple cube
Try to mill the surface
Check the simulation --> works as expected

Change the start point to z = xx in order that xx is above the solid
Try to simulate --> object will cutted from the start point to the real first point where the mill should start to work.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Windows 7 0.18.16110 (Git)**
* **Updated:** 2/6/2021

sliptonic, 2022-03-21:
This doesn't appear to be a simulation problem.  I've reviewed the use of the simulator and it seems to be correctly showing the generated gcode.  However, the original post points to an incorrect and inconsistent use of StartPoint.

The following operations currently have a StartPoint property:
surface
slot
all PathArea based ops (profile, pocket, millface, 3Dpocket)

With these ops, if the startpoint is set to 100,100,100,  the generated gcode NEVER goes to that point.  
For the PathArea ops, it first goes to 0,0,100, then rapids to 100,100,clearanceheight.

For slot, it goes directly to 100,100,clearanceheight.

First,  it's unclear (at least to me) what the purpose of the startpoint is.   It seems to be to force the path to first go to a known place, and then proceed from there.  However, if that is desired behavior, it should do so for all operations.  

Second, if we actually desire this feature,  the correct and safe behavior would be:
G0  Z<startpoint.z>
G0 X<startpoint.x>  Y<startpoint.y>  Z<startpoint.z>
...

luzpaz, 2022-03-21:
CC @PLChris (since you've also participated in the ticket's dedicated forum thread)

sliptonic, 2023-08-12:
Stale isssue.  Closing due to lack of engagment.  Reopen and update if there's still a bug to address

closed 2023-08-12 by sliptonic