FreeCAD/FreeCAD #5937: zoom unstable

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3994

* **Reporter:** RICCARDO
* **Date submitted:** 5/30/2019
* **FreeCAD version:** 0.18
* **Category:** Bug
* **Status:** feedback
* **Tags:** 

# Original report text

cambia visuale in automatico, sparisce tutto dopo qualche secondo, perde il centro schermo

English:
changes view automatically, disappears after a few seconds, loses the screen center

# Additional information

OS: Windows 7
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16110 (Git)
Build type: Release
Branch: (HEAD detached at upstream/releases/FreeCAD-0-18)
Hash: f7dccfaa909e5b9da26bf50c4a22ccca9bb10c40
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: Italian/Italy (it_IT)

# Steps to reproduce

aprire file esempio
fare scorrere il mouse sullo schermo
si nota la scomparsa dell'immagine, e la rotazione parziale del Gestore visuale 3d in alto a dx

English:
open example files
slide the mouse on the screen
note the disappearance of the image, and the partial rotation of the visual 3D manager at the top right

# Other bug information

* **Priority:** urgent
* **Severity:** crash
* **Category:** Bug
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2019-06-17 06:46
RICCARDO Can you reproduce this on 0.19_pre ?
Also please follow our guidelines (big yellow banner at the top of this page). Please also use english language in the bug reports. Thank you

---

### Comment by Kunda1 2019-08-22 20:31
Forum thread: https://forum.freecadweb.org/viewtopic.php?f=8&t=38759

---

### Comment by Kunda1 2019-08-23 00:22
RICCARDO no example files were attached.

BobScheid, 2023-04-15:
I believe I've stumbled upon this myself. The specific cause is unclear, but in sketcher, I've been trying to model a cross-section that I can then turn into a revolute. Doing the drawing is usually stable, doing the constraints is usually stable, but when center-clicking to move around and/or zooming out with mousewheel, the GUI will freeze. The only solution I've found is to force-close the program (it's still unresponsive after 20+ minutes), re-open, and hope I can avoid it. 

I have a feeling, but can't prove, that it has to do with the way the focus-vertex line of the parabolic sections are what's causing the freeze. I've recreated this part a few times and the problems seem to start when the parabolae enter the mix. I also think that center-click-and-drag while those lines are on screen might cause instability, but the zoom is guaranteed to do it.

**Steps to reproduce:** 
Open attached file
Edit sketch (double-clicking icon on left)
Small moves with center-click-and-drag should be OK
Zooming out two swipes on the mousewheel ought to be enough to freeze it

**My system**
Windows 11 Home 22H2
CPU Ryzen 5 5600G
32GB RAM
Radeon RX6600XT

**From FreeCAD's About:**
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

[mad_madea.zip](https://github.com/FreeCAD/FreeCAD/files/11240812/mad_madea.zip)

luzpaz, 2023-04-15:
Hi, thanks for the feedback! Can you kindly recreate in the development build ? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

BobScheid, 2023-04-16:
I was able to recreate it on that build. I was using the same file as the one provided in the ZIP earlier, identical in all respects, and the only difference to the process was that I drag-dropped the FCStd file into the window to open it instead of using the File > Open Recent menu.
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32552 (Git)
Build type: Release
Branch: master
Hash: e2b3a5d2f791fb78a61daac9bdab374058f682f2
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

I'm including a few extra pictures. First is a new migration message saying parabolas are being treated differently and won't be backward compatible. Second is an overall view just as the freeze happens, which hopefully can give you an idea of where it happens, like, at what scale the zoomout causes the crash. I also confirmed that the file's default zoom makes it less likely to crash, including when parabolae vertex/focus lines are on screen. I don't think it completely fixes it, just that it's not as likely at the closer zoom. 

[image]
[image]

I also wanted to point out that moving and zooming when not in sketch view, but when the profile is on the screen in the basic view, before getting into the sketcher, it seems like you can zoom and move with impunity. It's only the sketcher view that has the problem. 

Let me know if you need anything else. Thanks!

MadsonFerrari, 2023-05-31:
Hi Guys, 

I think that I have the same problem. I was trying to draw a lid for a plastic bowl and when I enter the sketch to draw the thread fillet the Freecad crashes during the zoom-out operation. I found this problem on versions 0.20.1 and 0.20.2. Every time I zoom out the crash occurs.

The crash occurs only inside the sketcher. 

Steps to reproduce the error: Open the attached drawing (zip file attached), enter in the sketc001, and zoom out.

[lid for a plastic bowl Sample.zip](https://github.com/FreeCAD/FreeCAD/files/11618592/lid.for.a.plastic.bowl.Sample.zip)

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Portuguese/Brazil (pt_BR)
[/code]
hardware info:
Mainboard Asus P8H61-M LX2 R2
Core: I5 3330
GPU: Raden RX550 4gb - No overclocking
Memory: 16GB DDR3 Dual Channel

luzpaz, 2023-06-01:
@Jolbas could you weigh-in on this 'zoom' related ticket as well (if you don't mind) ?

maxwxyz, 2024-02-26:
Is this still relevant? I cannot reproduce this behavior, also not https://github.com/FreeCAD/FreeCAD/issues/6234

luzpaz, 2024-03-02:
@BobScheid  @MadsonFerrari  can you still reproduce on latest stable or development ?

maxwxyz, 2024-03-24:
I'll close this as https://github.com/FreeCAD/FreeCAD/issues/6234 was also solved and there is no more feedback.

closed 2024-03-24 by maxwxyz