FreeCAD/FreeCAD #5938: Trackpad on Macbook unusable in any mouse model so far

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3995

* **Reporter:** robotarmy
* **Date submitted:** 5/31/2019
* **FreeCAD version:** 0.18
* **Category:** Bug
* **Status:** new
* **Tags:** Bountysource Potential, gesture, macOS, trackpad

# Original report text

Hi all, 

My trackpad and FreeCAD don't get along :( 

None of the two-finger things (pan, pinch to zoom) behave as I would expect, and it makes it almost impossible to use FreeCAD on my normal platform. 

Forum posts related: 
https://forum.freecadweb.org/viewtopic.php?f=3&t=34351
('Gesture mode' as is suggested doesn't work at all well for me)

https://forum.freecadweb.org/viewtopic.php?f=3&t=33282&start=10
(If lack of mac is the root cause, I'd be happy to loan one! Good luck DeepSOIC/all)

I think this may also be related to #2437 but that's listed as a 'minor feature request' on 0.15. This is a showstopper for me using FreeCAD. 

I've captured two videos with the default settings, and I'm not seeing what I would consider 'expected behaviour' with my trackpad, and none of the mouse models make it much more useful. 

You can see what I see here with only very minor finger movements in each case. 
https://www.youtube.com/watch?v=iA9-QKA6iQ8
https://www.youtube.com/watch?v=6hmtKJJUAKM

It 'should' behave (I suppose?) as described here: https://developer.apple.com/design/human-interface-guidelines/macos/user-interaction/mouse-and-trackpad/

Maybe setting some zoom multiplier would help it not be so wild, but the gestures are being interpreted incorrectly, as well as being too fast. 

The 'Gesture' model looks the most correct on the mouseover, but it seems that the two finger gestures are being picked up as scrolling (or something?) rather than the gestures they are. 

Thanks for all your hard work crew, it's come a long way!!

# Additional information

OS: macOS Sierra (10.12)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16117 (Git)
Build type: Release
Branch: (HEAD detached at 0.18.2)
Hash: dbb4cc6415bac848a294f03b80f65e888d531742
Python version: 3.6.7
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedKingdom (en_GB)

# Steps to reproduce

Run 0.18 on a Macbook with the mouse model set to 'Gesture' mode and try to do two-finger gestures. 

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2019-06-17 06:23
robotarmy please participate in https://forum.freecadweb.org/viewtopic.php?f=9&t=36387 
DeepSOIC has updated the gesture navigation code in 0.19_pre but mostly for windows. We need MacOS X testers that can give him feedback (because he doesn't own a mac).

---

### Comment by krite 2019-10-25 15:07
This issue is exactly as bad in Ubuntu on a Lenovo T480s, totally unusable.
The behaviour is also identical in the newest 19_pre from Sep 7.

Further analysis actually show that the scroll upp/down on the trackpad behaves as pressing pgup/pgdown-keys. Thus not zooming like a mouse at all, which is totally unusable of course.

The two-finger up/down-scroll is thus not usable (large hops), and the pinch-zoom is actually intercepted incorrectly and interfer with the two-finger action, and does not work at all.

---

### Comment by Kunda1 2020-09-16 13:07
DeepSOIC were you working on this ?

---

### Comment by DeepSOIC 2020-09-16 17:15
i am not. Fixing gigantic zoom sensitivity ( https://forum.freecadweb.org/viewtopic.php?f=9&t=49680 ) should have made things a lot better, but i don't have a macbook to try (and thus i can't work on it). chrisb ?

---

### Comment by DeepSOIC 2020-09-16 17:17
As for the linux part, I can try. I don't have linux, but i can install one in theory.

luzpaz, 2022-12-31:
@robot-army is this your original issue?

robot-army, 2023-01-01:
Yep! I haven't tested on any nightlies or anything, but...I think what was needed was a suitably capable developer with suitable hardware...it's not the sort of thing that would be easy to test "remotely" or whatever, I don't think. Could be wrong tho? Can I help?

luzpaz, 2023-01-01:
Just checking :)  
Thanks for the reply.

luzpaz, 2024-08-28:
This is safe to close, right @johnocfii ?

luzpaz, 2024-08-28:
It's time we add gesture support: https://doc.qt.io/qt-6.5/qnativegestureevent.html

luzpaz, 2024-09-26:
@hyarion just our of curiosity, could you weigh-in what it would take to implement this feature code-wise ? TIA

hyarion, 2024-09-26:
@luzpaz

Apple's says this about gestures:
> [With a Multi-Touch trackpad or Magic Mouse, you can tap, swipe, pinch, or spread one or more fingers to perform useful actions.](https://support.apple.com/en-us/102482)

What I think mac users would like to have here is the multi-touch navigation, similar to how you interact with multi-touch applications on an iPhone/iPad or Android device.

Maybe this is the same thing, but Apple tends to define their own meaning to stuff (like calling backspace delete *sigh*)

I have a bad feeling that this might be a place where it is difficult to get a good general solution that works well on all platforms and that more types of touch devices needs to be supported in windows/linux than on macos.

I think I would do as follows:
1. creating a separate code path for macos
2. make sure that zoom using two-finger-scrolling is disabled where multi-touch navigation should be used (enabled now)
3. pinch using twofingers to zoom
4. slide two fingers to rotate in 3d view (keep the setting for orbital vs turntable)
5. slide two fingers to pan in 2d view like sketcher (the reasoning here is that it should mimic how to navigate in a 2d application)
6. slide three fingers to pan in 3d view (maybe also allow this in 2d views)
7. rotating two fingers should either A) not tilt by default, or B) Have a spring like tensioning feeling which snaps it to the closest 45/90 degrees. (A modifier key could allow it to rotate freely).

Combining zoom, pan, and tilt at the same time should also work be supported. 

Code-wise I have this feeling that to get a good apple-y feeling to it, we might need to keep the mac code separate from the windows/linux gesture code.
It also suspect that Qt's gestures aren't good enough and that we need to implement this on a lower level (looking at the touch positions). This is especially true if we want to have a really nice touch screen experience (with wacom's cintiq or alike).

Maybe we could create a new a new navigation mode "Multi-touch" and disable the Gesture mode on mac?

Note though, above is pure speculations and my personal opinions and I could be missing stuff or being dead wrong.
I've worked with multi-touch input for mobile games but I haven't done any multi-touch coding in Qt or FreeCAD's navigation system.

hyarion, 2024-09-26:
> Just as an aside, gesture support in Fusion for the Mac is not great. I tend to avoid it when I use Fusion. This might be a hill too far for QT- based FreeCAD.

It seems like it possible to get the touch data from: https://doc.qt.io/qt-5/qtouchevent-touchpoint.html#details

luzpaz, 2024-09-27:
Can anyone think of another FOSS Qt based software that has formidable gesture support ?

prokoudine, 2024-09-27:
Hola. I can't think of any CAD programs that are both FOSS, Qt-based, and support pinch zoom et al.

I know that Qt6 does have some support for gestures (unsure about Qt5): https://doc.qt.io/qt-6/gestures-overview.html. However gestures are supposed to happen in the 3D view here, and navigation styles seem to largely depend on the OpenInventor API — as much as I can figure out from squinting at the source code (not much at all). Qt doesn't seem to be involved. But don't take my word for it.

hyarion, 2024-09-27:
Yes it is using the openinventor api, but the stack is: native->qt->quarter->coin3d->freecad.
In the discussion I just assumed that this is an issue we can solve, like either implementing touch in quarter & coin3d or ditching them and use qt directly for navigation.

luzpaz, 2024-10-05:
> but the stack is: native->qt->quarter->coin3d->freecad.
In the discussion I just assumed that this is an issue we can solve, like either implementing touch in quarter & coin3d or ditching them and use qt directly for navigation.

In that case we need to open upstream quarter and coin3d tickets? (to ascertain how to go about the final approach?)

luzpaz, 2024-10-06:
@hyarion legend :saluting_face: 

Can you share your working repo for others interested in following progress and also potentially collaborating ?

hyarion, 2024-10-06:
Sure! This is WIP and nothing works yet 🚧. I'm basically just trying to tunnel multi-touch events from qt, via the help of quarter, through coin and then handle it in a new MultitouchNavigator. 

touch coin branch (create classes for touch events that can be sent through the system):
https://github.com/hyarion/coin/tree/touchevent-support

freecad branch (quarter files are translating the qtouchevents to coin touchevents):
https://github.com/hyarion/FreeCAD/tree/multitouch-navigator

Gabriel29, 2024-12-21:
Hello, everybody!

I reached this page because trackpad gestures do not work my macOS. I integrated gestures 3 years ago on a propietary CAD software. At that time Qt did not offer a great solution for Gestures and I went with Native Gestures from the macOS API.

@hyarion did you manage to get things working for FreeCAD? Probably, I can help out with this features if I have enough time.

hyarion, 2024-12-21:
> Hello, everybody!
>
> I reached this page because trackpad gestures do not work my macOS. I integrated gestures 3 years ago on a propietary CAD software. At that time Qt did not offer a great solution [...]

Please go ahead. You can find my wip work in the comment above.
I think it would be preferred if Qt input was used and passed through coin3d using the openinventor api, but I won't dictate what to do as anything is better than nothing and I'm not currently working on it.

Feel free to pop in on discord in the developer channel if you like to chat about it :)

hunterirving, 2025-01-27:
Just chiming in to say that this is a feature I would like to see and would be happy to help test! I have a MacBook Pro and a Magic Trackpad.

luzpaz, 2025-01-28:
@Gabriel29 you still interested?

hyarion, 2025-01-28:
> [@Gabriel29](https://github.com/Gabriel29) you still interested?

Gabriel29 popped in on discord to chat about it a week ago, so I think he is still interested :)

Abdullah-R, 2025-03-12:
@hyarion @Gabriel29 Has there been a lot of progress on this? I found this issue after experiencing the lack of MacOS native trackpad gestures myself. I have quite a bit of downtime and would love to help out on building this feature.

hyarion, 2025-03-12:
> @hyarion @Gabriel29 Has there been a lot of progress on this? I found this issue after experiencing the lack of MacOS native trackpad gestures myself. I have quite a bit of downtime and would love to help out on building this feature.

Nope, I haven't heard anything from Gabriel either.

I think it would be wonderful if you'd like to look into it, and if Gabriel is still interested I'm sure you guys can find a way to cooperate on it :)

prokoudine, 2025-06-17:
> Do you have a bounty-system, to chip in?

Not at the moment.

ronangrant, 2025-10-03:
I've been experimenting with getting trackpad gestures working properly on macOS. Things like pinch to zoom and two-finger drag for orbit/pan. Here’s what I’ve found while poking around in a local Qt 6 build.

**What actually happens**

Pinch works when you use Qt 6’s `QNativeGestureEvent`. The problem is that macOS often sends rotation events alongside the pinch. That makes zoom feel a little wonky, but filtering those out makes it much smoother.

Two-finger drag is trickier. It doesn’t come through as a gesture at all. macOS just sends it as a `QWheelEvent` (high-res scroll). That’s why the current gesture navigation modes can’t really do anything useful with it.

**Current state**

Right now I have pinch to zoom working locally with rotation filtered out. It feels pretty natural.

Two-finger drag still just ends up as wheel-zoom instead of orbit/pan because the system doesn’t report it as a trackpad gesture.

**Side note on the existing "Touchpad" mode**

This might just be me, but the current "Touchpad" mode feels a little misleading. It doesn’t actually use trackpad gestures. It’s just a modifier-based scheme (Shift = pan, Alt = rotate, Ctrl+Shift = zoom). That’s totally useful for mouseless workflows, but the name suggests it’s for trackpad gestures when it’s really a keyboard/navigation style. Maybe something like “Keyboard” or “Modifier-based” would be clearer.

**What I’m thinking**

The safest approach might be to leave the existing input modes alone and add a new one that:

* Uses native pinch for zoom (already working)
* Interprets two-finger drag as orbit in the 3D view, but only when the event comes from a trackpad (`device()->type()==Touchpad`). That way mouse wheels and menus keep behaving normally.
* Maybe Shift + two-finger drag = pan

This would only affect the viewport when an actual trackpad is detected, so mouse users and other platforms stay safe.

That said, trackpad mappings are pretty subjective. I don’t want to just go ahead and build something without checking in, since it could mess up existing workflows or go against what people expect.

Curious to hear what others think.

luzpaz, 2025-10-03:
@JohnOCFII @ipatch   
Who else is on macOS that can weigh-in ?

hunterirving, 2026-08-20:
> > [@JohnOCFII](https://github.com/JohnOCFII) [@ipatch](https://github.com/ipatch) Who else is on macOS that can weigh-in ?
> > Hi, I've just started using freecad on a macbook, happy to test any updates to this issue.

I'm on macOS and would be happy to test (would love to see this feature in FreeCAD), but I don't know how to put a build together.

pavliha, 2026-08-21:
@hunterirving I've put a build with a fix that not yet merged. I will be happy if someone other than me willing to try it  https://github.com/pavliha/FreeCAD/releases/tag/release-all-platforms-2026.08.19

[image]

closed 2026-09-21 by chennes

hunterirving, 2026-10-03:
@pavliha, I finally got around to trying this (via the weekly build), and it works great! A huge improvement for Mac users. Thanks to everyone involved for getting this one merged!