FreeCAD/FreeCAD #5941: Toolbars get disabled on file open if 'Close & switch on opening file' is enabled

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3999

* **Reporter:** ifohancroft
* **Date submitted:** 6/2/2019
* **FreeCAD version:** 0.18
* **Category:** Bug
* **Status:** new
* **Tags:** file, options, preferences, start, toolbar, UI, workbench

# Original report text

***Summary:***

If you set FreeCAD's option 'Switch workbench after loading' to 'Part Design' and enable the option 'Close & switch on opening file', when you open a FreeCAD project file by double clicking on the file, when FreeCAD starts, the toolbars 'File, 'Workbench', 'Macro', 'View', 'Structure' and 'Navigation' wil be disabled.

You can see the toolbars for a second when FreeCAD starts and then dissapear (the whole second row of toolbars and those that are on the first row, dissapear as well. Tested with more than one FreeCAD project file.

***System info:***
Fedora 29 64 bit with XFCE as a desktop environment
uname -a output: Linux localhost.localdomain 5.0.17-200.fc29.x86_64 #1 SMP Mon May 20 15:39:10 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

# Additional information

OS: "Fedora release 29 (Twenty Nine)"
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.Unknown (0.18.2 according to the Fedora repo package info)
Build type: Unknown
Python version: 3.7.3
Qt version: 4.8.7
Coin version: 3.1.3
OCC version: 6.9.1.oce-0.18
Locale: English/UnitedStates (en_US)
(It was installed from the Fedora repo via DNF)

# Steps to reproduce

0. Backup your FreeCAD config files to not lose your current configuration
1. Open FreeCAD
2. Go to Edit -> Preferences -> Start
3. Change 'Switch workbench after loading' from 'Start' to 'Part Design' (haven't tested if selecting a workbench different than 'Part Design' makes a difference
4. Enable the option 'Close & switch on opening file' (this is irregardless of whether the option 'Close start page after loading' is enabled or not)
5. Hit ok (or apply and ok, it doesn't matter)
6. Close FreeCAD
7. Find a FreeCAD project file in your files using your file manager and double click it (or I guess single click if you have set your file manager to use single click for file opening)
8. You will see the toolbars in question for a second and then you will see them dissapear. (Now, since the Navigation toolbar isn't usually present/enabled in Part Design anyway, you can verify that it also dissapears as well as you first check the Start workbench and see it is enabled there, if after you reproduce the bug you go to Start again, you will see it is also one of the disabled toolbars)
9. To reproduce the bug again or to reset FreeCAD's enabled toolbars, delete everything in both $HOME/.FreeCAD and $HOME/.config/FreeCAD (haven't tested which config files keeps the toolbars options and the Start options) Keep in mind that deleting those files will reset every FreeCAD setting for your user.

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **OS: Fedora 29**
* **Platform:** x86_64
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2019-06-02 12:33
Thank you for the ticket. Some orientation on how to use our bugtracker: read the giant yellow banner at the top of the page. We ask you to open a forum thread first to discuss your issue on the forum before opening a ticket. Many issues are solved this way. Also please do not create tags if they don't exist. 

Now please create a forum thread and cross link ticket<->thread so we can track this issue's progress. Thank you for your comprehension.

FYI, you are using an out of date version of 0.18 that isn't py3/qt5 and has obsolete dependencies.

---

### Comment by ifohancroft 2019-06-03 01:18
I am sorry. I did read the How to submit bugs but thought making a post is only required for feature requests.
I had a checkmark on 'Existing tags' so I thought that it wouldn't let me set a tag if it doesn't exist. I will create a thread as soon as I get on my pc.
The version is the latest from the Fedora repo and it was in-fact updated the other day but I am after all one Fedora version behind. I will look into building FreeCAD from source.
[Forum post](https://forum.freecadweb.org/viewtopic.php?f=3&t=36785)
P.S. Bug still present in FreeCAD 0.19 compiled from git following the out of source build compile guide in [CompileOnUnix](https://www.freecadweb.org/wiki/CompileOnUnix)

OS: "Fedora release 29 (Twenty Nine)" (XFCE/xfce)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.16886 (Git)
Build type: Unknown
Branch: master
Hash: ed47e962d2c821bf1792889f6d7bdf457dcf6c9e
Python version: 2.7.15
Qt version: 4.8.7
Coin version: 3.1.3
OCC version: 6.9.1.oce-0.18
Locale: English/UnitedStates (en_US)

P.P.S. In the ticket description, where I've written 'Test with more than one FreeCAD project file.' it should be 'Tested with more than one FreeCAD project file.'

---

### Comment by Kunda1 2019-06-17 06:12
ifohancroft so is this still an issue because you were using python2 instead of py3 ?

---

### Comment by ifohancroft 2019-06-18 17:05
Kunda1 Yes. It is still a problem, even with the Appimage:

OS: Fedora 29 (Twenty Nine)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16117 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.18.2)
Hash: dbb4cc6415bac848a294f03b80f65e888d531742
Python version: 3.6.7
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

luzpaz, 2023-04-15:
@ifohancroft hi, sorry for delay on this. Can I ask you to kindly retest and see if this is still an issue on latest development build (v0.21.xxxxx)?

yorikvanhavre, 2023-04-19:
I just tested (using BIM Wb instead of PartDesign) and no toolbar is vanishing...

ifohancroft, 2023-04-19:
> @ifohancroft hi, sorry for delay on this. Can I ask you to kindly retest and see if this is still an issue on latest development build (v0.21.xxxxx)?

Hi. Of course! I'll try to test it later today.

adrianinsaval, 2023-05-11:
I can reproduce and it seems even worse than described:
[image]
```yml
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.33110 (Git)
Build type: Release
Branch: master
Hash: 18e40941ceb36c52701a7c74ea26043ef3385701
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C) [ OS: Spanish/Paraguay (es_PY) ]
Installed mods: 
  * AirPlaneDesign 0.4.0
  * Assembly3 0.11.4
  * Assembly4 0.50.1
  * CfdOF 0.6.8
  * CurvedShapes 1.0.4
  * Curves 0.6.8
  * Defeaturing
  * DynamicData 2.46.0
  * fasteners 0.4.56
  * FCsuWB-dev
  * Glass
  * Plot 2022.4.17
  * SelectorToolbar
  * Silk 0.1.1
```

ifohancroft, 2023-05-12:
> @ifohancroft hi, sorry for delay on this. Can I ask you to kindly retest and see if this is still an issue on latest development build (v0.21.xxxxx)?

So sorry for not testing yet. Trying to fit too much stuff into a single day doesn't seem to work that well in practice :D I'll try to test it before the weekend ends.

FEA-eng, 2026-10-03:
See [this](https://github.com/FreeCAD/FreeCAD/issues/5941#issuecomment-4124604577)

flaviut, 2026-11-02:
The relevant code here would have been `checkPostOpenStartPage()`, which has been removed with the Start workbench in d0a87d3e0c & cfb0873d38

That said, I've discovered (from my clanker) that `!isVisible()` and `isHidden()` mean different things when it comes to parent objects being visible/not visible:

 - `isHidden()`: was this widget itself directly hidden?
 - `isVisible()`: are this widget and its ancestors shown?

So that was likely why this issue came up in the first place.

closed 2026-11-02 by flaviut