FreeCAD/FreeCAD #5942: [Feature Request] Revit rotation stop on Scroll Release

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4002

* **Reporter:** ifohancroft
* **Date submitted:** 6/6/2019
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** keyboard, mouse, navigation, rotate

# Original report text

Currently, rotation in Revit navigation style works the following way:

1.    To rotate, start moving the mouse while holding Shift+Scroll.
2.    If you release just Shift and you keep moving - you are no longer rotating. You are panning.
3.    If you release just the Scroll and you keep moving - you are still rotating.
4.    To stop rotating, you need to release both Shift and Scroll.

I am coming from Blender and while the Revit mode works the way I have setup my Blender config, the details above confuse me very much and using both FreeCAD and Blender it would feel more comfortable for them to have them work the same way.

Here's what changes I am suggesting to the steps above and why they would be helpful in practice:

1.    To rotate, start moving the mouse while holding Shift+Scroll <- This one stays the same.
2.    If you release just Shift and you keep moving - you are still rotating. <- No need to having to hold shift forever once you've entered rotation mode. You can let go of it.
3.    If you release just the Scroll and you keep moving - you are moving the mouse freely (no rotation, no panning) <- This is the main idea and main change I would like to see. The idea behind is so that you can readjust your mouse and by pressing scroll again you continue your rotation.
4.    To stop rotating, you need to release both Shift and Scroll <- This is like it currently is, also it is like the proposed #3 but then you don't press scroll again and you let go of Shift so it's not really a separate step in a way.

TL;DR: When you release just Scroll during rotation, you should be able to move the mouse freely instead of continue to be in rotation mode.
Ideally, also: When you release just Shift, you should still be in rotation mode.

I think No.2 from the proposed changes (being able to still rotate while letting go of Shift) may affect the following:
Currently, if you are panning and you start holding shift, now you are rotating. I believe implementing #2 from the suggested changes may stop this form working this way.

[Forum post](https://forum.freecadweb.org/viewtopic.php?f=8&t=36830)

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by ifohancroft 2019-06-06 00:27
Above, where it says #3 or #02 that's meant to mean point 2 or 3 of the points mentioned. (Maybe we need the ability to edit our tickets after we've posted them.

luzpaz, 2022-11-17:
CC @ifohancroft

donovaly, 2022-11-21:
I backported this also for FreeCAD 0.20.2

closed 2022-11-21 by donovaly