FreeCAD/FreeCAD #5945: Build number & hash are no more available on daily build (Linux)

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4014

* **Reporter:** openBrain
* **Date submitted:** 6/10/2019
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** assigned
* **Tags:** 

# Original report text

Build number & hash that identifies FreeCAD release are no more available in Linux daily builds.
Windows builds seems to not have the same problem.
---
[s][Forum thread (FR)](https://forum.freecadweb.org/viewtopic.php?f=12&t=36578)[/s]

[Dedicated thread opened](https://forum.freecadweb.org/viewtopic.php?f=42&t=37216)

# Additional information

Example with latest daily build as of today :
```OS: Ubuntu 18.04.2 LTS (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.
Build type: Release
Python version: 3.6.7
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: French/France (fr_FR)```

# Steps to reproduce

1. Open FreeCAD
2. Click on "Help" -> "About FreeCAD" -> "Copy to clipboard"
3. Paste anywhere you want

# Other bug information

* **Priority:** immediate
* **Severity:** major
* **Category:** Bug
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by openBrain 2019-06-10 12:14
I tagged it with high importance/priority as it's a really useful function at forum discussions.

---

### Comment by Kunda1 2019-06-10 17:34
Punting to sgrogan

---

### Comment by openBrain 2019-06-22 16:27
Priority --> Urgent according [this thread](https://forum.freecadweb.org/viewtopic.php?f=3&t=37186#p316103)

---

### Comment by Kunda1 2019-06-22 18:49
kkremitzki Can this be fast tracked ?

---

### Comment by kkremitzki 2019-06-23 05:27
Kunda1 I don't have the bandwidth to try to find a solution for this right now, but I am keeping an eye on the issue to help implement it once a solution is found, or if anyone has questions about the current build automation.

sgrogan, 2022-02-07:
I think this cna be closed as solved. [https://github.com/FreeCAD/FreeCAD/releases/tag/0.19.3](url)

luzpaz, 2022-05-27:
@adrianinsaval I don't think this is solved, right?

adrianinsaval, 2022-05-28:
Is this about the PPA? If so I think not but I don't have an Ubuntu at hand to confirm. This happened because the compilation doesn't use a full clone of the repo, it might be solvable by switching to the source package from the weekly-builds page which is generated after preprocessing the version information. Another option is to include a patch for Version.h.cmake that shall be updated every day before building to have the correct info.
@kkremitzki 

Appimage is not affected by the flatpak also has incorrect info displayed for the same reason. I will be working with the flatpak maintainer to fix it there.

0penBrain, 2022-05-28:
@adrianinsaval I think most answers to your current questions are in the thread linked in the description. ;)

adrianinsaval, 2024-01-26:
Ubuntu ppa is now using sources as generated by https://github.com/FreeCAD/FreeCAD-Bundle/commit/f2b7d251cf4765e7d656730bc5236569d293b739 and https://github.com/FreeCAD/FreeCAD-Bundle/commit/2d3a76d348a6a7fc2d7625b455ea98e07d67cc94 so they should include the correct revision number

closed 2024-01-26 by adrianinsaval