FreeCAD/FreeCAD #5950: Crash upon solving expression referencing named Constraint

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4031

* **Reporter:** dhka2
* **Date submitted:** 6/20/2019
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** confirmed
* **Tags:** 

# Original report text

Freecad git build as of 18.06.2019 sometimes crashes with signal 11 upon referencing a named constraint in an expression. This is width python 3 and qt5.

stdout output:

```Program received signal SIGSEGV, Segmentation fault.
#0  /lib/libc.so.6(+0x378b0) [0x7f160e65a8b0]
#1  0x7f160fe90f76 in App::ObjectIdentifier::Component::get(Py::Object const&) const from /usr/lib/freecad/lib/libFreeCADApp.so+0x1c6
#2  0x7f160fe96638 in App::ObjectIdentifier::access(App::ObjectIdentifier::ResolveResults const&, Py::Object*) const from /usr/lib/freecad/lib/libFreeCADApp.so+0x1b8
#3  0x7f160fe9ae14 in App::ObjectIdentifier::getPyValue(bool, bool*) const from /usr/lib/freecad/lib/libFreeCADApp.so+0x84
#4  0x7f160fe04ca2 in App::VariableExpression::_getPyValue(int*) const from /usr/lib/freecad/lib/libFreeCADApp.so+0x62
#5  0x7f160fdf3e2c in App::Expression::getPyValue(int, int*) const from /usr/lib/freecad/lib/libFreeCADApp.so+0x4c
#6  0x7f160fdf4f62 in App::Expression::eval(int) const from /usr/lib/freecad/lib/libFreeCADApp.so+0x42
#7  0x7f16105e2cfc in Gui::Dialog::DlgExpressionInput::textChanged(QString const&) from /usr/lib/freecad/lib/libFreeCADGui.so+0x1fc
#8  0x7f160ec388e7 in QMetaObject::activate(QObject*, int, int, void**) from /lib/libQt5Core.so.5+0x777
#9  0x7f160f15c963 in QLineEdit::textChanged(QString const&) from /lib/libQt5Widgets.so.5+0x33
#10  /lib/libQt5Widgets.so.5(+0x29c662) [0x7f160f163662]
#11  0x7f160ec388e7 in QMetaObject::activate(QObject*, int, int, void**) from /lib/libQt5Core.so.5+0x777
#12  0x7f160f164276 in QWidgetLineControl::textChanged(QString const&) from /lib/libQt5Widgets.so.5+0x36
#13  0x7f160f167710 in QWidgetLineControl::finishChange(int, bool, bool) from /lib/libQt5Widgets.so.5+0x2a0
#14  0x7f160f16abaf in QWidgetLineControl::processKeyEvent(QKeyEvent*) from /lib/libQt5Widgets.so.5+0xebf
#15  0x7f160f15caeb in QLineEdit::keyPressEvent(QKeyEvent*) from /lib/libQt5Widgets.so.5+0x1b
#16  0x7f161053668f in Gui::ExpressionLineEdit::keyPressEvent(QKeyEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x1f
#17  0x7f160f060b88 in QWidget::event(QEvent*) from /lib/libQt5Widgets.so.5+0x258
#18  0x7f160f162ebb in QLineEdit::event(QEvent*) from /lib/libQt5Widgets.so.5+0xab
#19  0x7f160f01fda4 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/libQt5Widgets.so.5+0x84
#20  0x7f160f02833c in QApplication::notify(QObject*, QEvent*) from /lib/libQt5Widgets.so.5+0x118c
#21  0x7f1610539019 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x89
#22  0x7f160ec0def9 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/libQt5Core.so.5+0x179
#23  /lib/libQt5Widgets.so.5(+0x1b7b42) [0x7f160f07eb42]
#24  0x7f160f01fda4 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/libQt5Widgets.so.5+0x84
#25  0x7f160f0273c1 in QApplication::notify(QObject*, QEvent*) from /lib/libQt5Widgets.so.5+0x211
#26  0x7f1610539019 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x89
#27  0x7f160ec0def9 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/libQt5Core.so.5+0x179
#28  0x7f160ce9a50d in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /lib/libQt5Gui.so.5+0x16d
#29  0x7f160ce9f526 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) from /lib/libQt5Gui.so.5+0x116
#30  0x7f160ce7a55c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/libQt5Gui.so.5+0xbc
#31  /lib/libQt5XcbQpa.so.5(+0x6db7c) [0x7f16088f2b7c]
#32  /lib/libglib-2.0.so.0(g_main_context_dispatch+0x2bf) [0x7f160c2ca90f]
#33  /lib/libglib-2.0.so.0(+0x6d869) [0x7f160c2cc869]
#34  /lib/libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7f160c2cc8ae]
#35  0x7f160ec63079 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/libQt5Core.so.5+0x69
#36  0x7f160ec0cabc in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/libQt5Core.so.5+0x13c
#37  0x7f160f21864e in QDialog::exec() from /lib/libQt5Widgets.so.5+0x1de
#38  0x7f153a96cc92 in openEditDatumDialog(Sketcher::SketchObject*, int) from /usr/lib/freecad/lib/SketcherGui.so+0x422
#39  0x7f153a96d77f in finishDistanceConstraint(Gui::Command*, Sketcher::SketchObject*, bool) from /usr/lib/freecad/lib/SketcherGui.so+0x19f
#40  0x7f153a970f66 in CmdSketcherConstrainDistanceY::activated(int) from /usr/lib/freecad/lib/SketcherGui.so+0xb86
#41  0x7f161056a30c in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /usr/lib/freecad/lib/libFreeCADGui.so+0x36c
#42  0x7f160ec388e7 in QMetaObject::activate(QObject*, int, int, void**) from /lib/libQt5Core.so.5+0x777
#43  0x7f160f0195e3 in QAction::triggered(bool) from /lib/libQt5Widgets.so.5+0x43
#44  0x7f160f01bcca in QAction::activate(QAction::ActionEvent) from /lib/libQt5Widgets.so.5+0xfa
#45  0x7f160f01c5d5 in QAction::event(QEvent*) from /lib/libQt5Widgets.so.5+0x55
#46  0x7f160f01fda4 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/libQt5Widgets.so.5+0x84
#47  0x7f160f0273c1 in QApplication::notify(QObject*, QEvent*) from /lib/libQt5Widgets.so.5+0x211
#48  0x7f1610539019 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x89
#49  0x7f160ec0def9 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/libQt5Core.so.5+0x179
#50  0x7f160cec8890 in QShortcutMap::dispatchEvent(QKeyEvent*) from /lib/libQt5Gui.so.5+0x110
#51  0x7f160cec8973 in QShortcutMap::tryShortcut(QKeyEvent*) from /lib/libQt5Gui.so.5+0x63
#52  0x7f160ce7dedc in QWindowSystemInterface::handleShortcutEvent(QWindow*, unsigned long, int, QFlags<Qt::KeyboardModifier>, unsigned int, unsigned int, unsigned int, QString const&m /lib/libQt5Gui.so.5+0x18c
#53  0x7f160ce9a42e in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /lib/libQt5Gui.so.5+0x8e
[... 112 lines cut ...]

### Comment by mikeprice99 2020-07-02 11:54
Just a few observations:
1/ A vertical constraint does not take a parameter, so no expression button is available. Did you mean vertical distance constraint?
2/ 'Sketch' in your sample file does not include a constraint called 'w' (Did you mean Spreadsheet.w )?
3/ If I use a vertical distance constraint and enter the value Sketch002.Constraints.Nw (which does exist!) it works.

So I guess either 
a) the bug does not affect Windows 
b) it has been fixed since revision 18336
c) it is particular to branch makepkg

My test environment: 
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21769 (Git)
Build type: Release
Branch: master
Hash: 801bfe4ba5f0910e9b2885917c521ab7fd8e9020
Python version: 3.6.8
Qt version: 5.12.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/United Kingdom (en_GB)

adrianinsaval, 2022-02-10:
while this comes from the same issue as #5770 the crash should be adressed separately IMO, will test later in arch to see if it's still crashing

adrianinsaval, 2022-02-11:
No crash

OS: Linux 5.16.7-arch1-1 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.27465 (Git)
Build type: Release
Branch: master
Hash: 092afacfcb5127eb67c385a43cbe34a4cd15bef4
Python version: 3.10.2
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United Kingdom (en_GB)

but there was no named constrains in Sketch so I'm not fully grasping what was OP doing to begin with, naming a constrain w and then using it worked without issues so it might not have been a duplicate of the other issues to begin with.

There was a w cell in the spreadsheet though, but similarly it didn't give me any issues

adrianinsaval, 2023-04-14:
I'm closing this one due to not being reproducible and inactivity, if there ever was an actual problem it's likely been solved.

closed 2023-04-14 by adrianinsaval