FreeCAD/FreeCAD #5953: Easier Way to Toggle BOPCheck On / Off

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4034

* **Reporter:** bejant
* **Date submitted:** 6/24/2019
* **FreeCAD version:** 0.18
* **Category:** Feature
* **Status:** new
* **Tags:** UI

# Original report text

Provide an easier method to enable / disable BOPCheck instead of always having to go through Tools > Edit Parameters.  The initial thought was to provide a button or tickbox in the Check Geometry widget but (other than a progress indicator) the Check Geometry widget doesn't appear until after Check Geometry has arrived at the results.  Even though a new user would have to first run Check Geometry before seeing an option for BOPCheck enable / disable, the user would at least be alerted that the option exists, and simultaneously be presented with a simpler way to change the setting.

TheMarkster posted in the forum that he had written a tiny macro to quickly and easily enable/disable RunBOPCheck, which gave rise to this feature request.

Forum discussion initially started here, and this post includes the macro:
https://forum.freecadweb.org/viewtopic.php?f=3&t=37156&start=10#p316228

I later started a discussion for this feature request to avoid further hijacking the original thread:
https://forum.freecadweb.org/viewtopic.php?f=8&t=37240 

# Additional information

OS: Ubuntu 18.04.2 LTS
Word size of OS: 32-bit
Word size of FreeCAD: 32-bit
Version: 0.18.1.
Build type: Release
Python version: 3.6.7
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

# Other bug information

* **Priority:** low
* **Severity:** minor
* **Category:** Feature
* **OS: Ubuntu 18.04.2 LTS**
* **Platform:** 32 bit
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by TheMarkster 2019-06-25 01:59
https://github.com/FreeCAD/FreeCAD/pull/2291

---

### Comment by Kunda1 2019-09-30 00:07
TheMarkster FYI, please use the mantisBT syntax to remotely close a MantisBT ticket from a Github PR. And do you mind stating here how the PR addresses this ticket? Thanks. (I'll associate the commit with this ticket manually)

---

### Comment by Kunda1 2019-09-30 00:10
TheMarkster, Sorry, let me correct myself, you did try to use the syntax but didn't execute it correctly. You can omit the superflous 0's and just use the number after the pound sign in the future... so either 
Fixed #(issue number) <-closes  
or 
issue #(issue number) <-references

xtemp09, 2023-04-18:
If I understand it correctly, the issue is either solved or no longer relevant.

luzpaz, 2023-04-18:
Indeed. Credit to @mwganson!  
Thx @xtemp09 :+1: 
Closing.

closed 2023-04-18 by luzpaz