FreeCAD/FreeCAD #5958: lack of error message on failing to pad sketch

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4040

* **Reporter:** freman
* **Date submitted:** 6/27/2019
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

There is clearly some test or operation which is failing here but nothing is output at a stage where it can be useful. 

Having effectively broken the outline in the sketch, no error is produce on closing when the pad attempts to recompute. 

Errors should be adequately trapped and useful output provided as to what failed, at a level where that information is available. 

Similarly, on attempting to change the length of the pad in this condition, a window with the text "Error! is really not that informative.

This could be handled in a way which is more informative to the user. 

# Additional information

OS: Linux (LXDE/LXDE)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.17119 (Git)
Build type: Release
Branch: master
Hash: e72b48e0cfcf4773e4b6527c74f2bac11ccff1ab
Python version: 2.7.15
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedKingdom (en_GB)

# Steps to reproduce

Create a sketch with closed form.
pad from sketch.
edit sketch to no longer be a closed form.
close sketch.
Note exclamation on pad: try to find some error output, explaining the error. 

The attached file shows the state of a model after inadvertently leaving a sketch without a closed form ( the construction lines masked that there was a missing line to close the loop ). 

On exiting sketch editor, the pad element named base-block showed an exclamation mark but there was no output in report window nor on the console showing that an error had occurred. 

Subsequent attempts to edit the pad produce a tiny window with the text "Error!". As before, nothing more is produced. 

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by freman 2019-06-28 06:06
TheMarkster has noticed that the presence of the path elements seems to be eating the error context, leading to the terse and unhelpful "Error!".

https://forum.freecadweb.org/viewtopic.php?f=3&t=37313#p317383

I have confirmed that the presence of the jobs do not affect this but at least one path in either job will trigger this bug.  With no paths present there is a proper error msg which reports "failed-to-validate-broken-face".

Presumably the change to pad triggers a recompute of the paths and the error context is lost. It seems that this should be trapped and reported before attempting a recompute ( which will be futile anyway if the base pad object is broken).

mwganson, 2023-11-28:
> @mwganson can you still reproduce this ?

The file no longer functions in my version of FreeCAD, presumably due to changes in the Path workbench source code over the last 4 years.  At any rate, I get valid error messages about failing to validate broken face.

luzpaz, 2023-11-28:
Then looks like we should close ticket.

closed 2023-11-28 by luzpaz