FreeCAD/FreeCAD #5959: Navigation cube hardcoded to use Helvetica

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4041

* **Reporter:** leong
* **Date submitted:** 6/28/2019
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** new
* **Tags:** #lowhangingfruit

# Original report text

in the latest 17147 and in the build before it the navigation cube has lost its lettering.

# Additional information

OS: Mageia 6 (KDE//usr/share/xsessions/01plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.17147 (Git) AppImage
Build type: Release
Branch: master
Hash: 251a3b1ff5517d929b4993d8836e17e01ae2623c
Python version: 3.7.3
Qt version: 5.9.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

# Steps to reproduce

Happens everytime I open FreeCAD

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Linux Mageia 6**
* **Updated:** 12/29/2021

# Discussion from Mantis ticket

### Comment by openBrain 2019-06-30 14:28
leong as suggested by tag, could you please open a topic in the ["Help on using FreeCAD" subforum](https://forum.freecadweb.org/viewforum.php?f=3) ?
The ticket is vague and a community discussion will probably allow to narrow down the possible cause(s). ;)
Thanks

---

### Comment by leong 2019-06-30 17:25
OK. It appears this may be an issue with my system and some font change, but I cannot find it. Will do as you suggest/

---

### Comment by Kunda1 2019-07-01 01:00
Forum thread: https://forum.freecadweb.org/viewtopic.php?f=3&t=37401

---

### Comment by jmaustpc 2019-07-01 01:12
It works perfectly here in both PPA versions. I have a 10 year old laptop with an Nvidea GPU with open source drivers. We saw that "save picture" was not working with a combination of some QT versions and Intel GPU/Driver .... I wonder if this is also related to the QT version and Intel GPU/Driver?

OS: Ubuntu 18.04.2 LTS (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.
Build type: Release
Python version: 3.6.8
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/Australia (en_AU)

OS: Ubuntu 18.04.2 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.1.
Build type: Release
Python version: 3.6.7
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/Australia (en_AU)

---

### Comment by leong 2019-08-09 11:37
I posted answer on forum [FreeCAD forum](https://forum.freecadweb.org/viewtopic.php?f=3&t=37401&p=325867#p325867)
Basically the navigation cube uses Helvetica font and it is not on my system. Had to add it. 
Maybe the code should specify another font? that would fix the issue.

---

### Comment by chennes 2021-12-29 05:12
I've changed the title of the issue to reflect the actual bug here: we assume Helvetica is available on all systems, to be used as a Sans Serif font, but that is not a good assumption.

onekk, 2022-08-26:
Problem is that "Helvetica" is a generic name and fontconfig could have some strange ideas about what is the "helevetica" substitution font.

Maybe using some more widespread font like "DejaVuSans" or "Liberation" will be a good assumtption.as they are installed on most distribution.

Helvetica is an Adobe fonts, Arial is the MS equivalent, but remaining on TTF one of the cited font family will be a "fairly safe" assumption, eventually a check and some warning or "minmal requirement" will be a good thing.

Regards

Carlo D.

luzpaz, 2023-04-18:
Indeed. Should be closed. Thx!

closed 2023-04-18 by luzpaz