FreeCAD/FreeCAD #5960: Solver claims solvable sketch to be unsolved

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4042

* **Reporter:** chrisb
* **Date submitted:** 6/28/2019
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** assigned
* **Tags:** 

# Original report text

See [forum discussion](https://forum.freecadweb.org/viewtopic.php?f=3&t=37366)
The sketcher claims some sketches to be unsolvable, although they can be corrected manually to fulfill all constraints.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 2/6/2021

Zolko-123, 2022-10-25:
should be closed

luzpaz, 2022-10-26:
CC @PLChris

PLChris, 2022-12-09:
@luzpaz, why did you ping me? All information is there including an example.

luzpaz, 2022-12-17:
@PLChris sorry, but wasn't it you who opened original ticket on mantis?

luzpaz, 2023-12-01:
```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35249 (Git) AppImage
Build type: Release
Branch: main
Hash: 94ca51b7799b2ac60a2a56d7e6e753bb97a73671
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * OpenDark 2023.11.27
```

I get:

> 07:15:06  Updating geometry: Error build geometry(0): gp_Circ::SetRadius() - radius should be positive number
07:15:06  Invalid solution from DogLeg solver.
07:15:06  Updating geometry: Error build geometry(0): gp_Circ::SetRadius() - radius should be positive number
07:15:06  Invalid solution from LevenbergMarquardt solver.
07:15:06  Updating geometry: Error build geometry(0): gp_Circ::SetRadius() - radius should be positive number
07:15:06  Invalid solution from BFGS solver.
07:15:06  Updating geometry: Error build geometry(0): gp_Circ::SetRadius() - radius should be positive number
07:15:06  Invalid solution from SQP(augmented system) solver.
07:15:06  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Invalid constraint index: 1
07:15:06  solver_issue_cb_2#Sketch: App.getDocument('solver_issue_cb_2').getObject('Sketch').setDatum(1,App.Units.Quantity('50.000000 mm'))

closed 2024-12-29 by github-actions[bot]