FreeCAD/FreeCAD #5962: DogLeg does not consistently converge to a same solution given the same input.

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4045

* **Reporter:** abdullah
* **Date submitted:** 6/30/2019
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** assigned
* **Tags:** 

# Original report text

https://forum.freecadweb.org/viewtopic.php?f=3&t=37369&p=317988#p317988

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 2/6/2021

maxwxyz, 2024-12-21:
@AjinkyaDahale still relevant?

closed 2025-08-09 by github-actions[bot]

PaddleStroke, 2026-01-08:
@chennes could you please update the description of the ticket?

forum updated link: https://forum.freecad.org/viewtopic.php?t=37369&start=20

File : [20190629b-Symmetry.zip](https://github.com/user-attachments/files/24497612/20190629b-Symmetry.zip)

Reproduction steps : 

- Open 20190629b-Symmetry.FCStd.
- Open the Sketch for editing.
- Select the lower outer endpoint of the left triangle, then the Y-Axis, then the lower outer endpoint of the right triangle.
- Apply a Symmetry Constraint (press 'S' or use the toolbar).
- The Bug: The sketch will either "collapse" (shrink into points), or the DogLeg solver will fail, leaving the elements immobilized despite having degrees of freedom remaining.

I can confirm that this still happens in today's version.