FreeCAD/FreeCAD #5963: Implementation flaw in View/Show all objects

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4046

* **Reporter:** openBrain
* **Date submitted:** 7/2/2019
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** new
* **Tags:** 0.18, GUI

# Original report text

The command "Show all objects" found in View -> Visibility not always makes visible the last operation in a PartDesign Body.

---

[Forum thread](https://forum.freecadweb.org/viewtopic.php?f=3&t=37421)

# Additional information

```OS: Ubuntu 18.04.2 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.1.
Build type: Release
Python version: 2.7.15rc1
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: French/France (fr_FR)```

---

```OS: Ubuntu 18.04.2 LTS (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19. // 2019/06/29
Build type: Release
Python version: 3.6.8
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: French/France (fr_FR)```

---

```OS: Windows 8.1
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.17171 (Git)
Build type: Release
Branch: master
Hash: d19470a9711ea604f3ca6c93e46afadf64d5bb87
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: Bulgarian/Bulgaria (bg_BG)```

# Steps to reproduce

1. Open the attached file[*]Select View -> Visibility -> Show all objects => 'Draft' is incorrectly made visible instead of 'Pad003'

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 2/6/2021

luzpaz, 2023-11-11:
I can still reproduce on 
```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34959 (Git) AppImage
Build type: Release
Branch: main
Hash: 640018bfa7878a79e1b3c220ddd488d66464dc70
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
```

bgbsww, 2023-12-12:
I'm not sure this is a bug.  If you look at the console, you can see that all of the objects are getting Visibility set:
```
>>> ### Begin command Std_ShowObjects
>>> ## Gui/CommandView.cpp(1206)
>>> # Gui.getDocument("allVisibleBug").getObject("Body001").Visibility=True
>>> ## Gui/CommandView.cpp(1206)
>>> # Gui.getDocument("allVisibleBug").getObject("Origin002").Visibility=True
>>> ## Gui/CommandView.cpp(1206)
>>> # Gui.getDocument("allVisibleBug").getObject("X_Axis002").Visibility=True
>>> ## Gui/CommandView.cpp(1206)
>>> # Gui.getDocument("allVisibleBug").getObject("Y_Axis002").Visibility=True
>>> ## Gui/CommandView.cpp(1206)
>>> # Gui.getDocument("allVisibleBug").getObject("Z_Axis002").Visibility=True
>>> ## Gui/CommandView.cpp(1206)
>>> # Gui.getDocument("allVisibleBug").getObject("XY_Plane002").Visibility=True
>>> ## Gui/CommandView.cpp(1206)
>>> # Gui.getDocument("allVisibleBug").getObject("XZ_Plane002").Visibility=True
>>> ## Gui/CommandView.cpp(1206)
>>> # Gui.getDocument("allVisibleBug").getObject("YZ_Plane002").Visibility=True
>>> ## Gui/CommandView.cpp(1206)
>>> # Gui.getDocument("allVisibleBug").getObject("Sketch001").Visibility=True
>>> ## Gui/CommandView.cpp(1206)
>>> # Gui.getDocument("allVisibleBug").getObject("Pad002").Visibility=True
>>> ## Gui/CommandView.cpp(1206)
>>> # Gui.getDocument("allVisibleBug").getObject("Pad003").Visibility=True
>>> ## Gui/CommandView.cpp(1206)
>>> # Gui.getDocument("allVisibleBug").getObject("Draft").Visibility=True
>>> ### End command Std_ShowObjects
```

However, Pad002, Pad003, and Draft are all inside Body001.  When there are multiple objects in a Body, if you make one Visiblity it disables the others.  In this case, since Draft happens to be the last object in the Body, it gets set Visibility last, and thus it wins.

So, while this is confusing to the user, unless we change the behavior of Bodies, I don't think it can be addressed.

@luzpaz maybe this should not be in the 1.0 Milestone?  Or a different kind of description of the problem.

maxwxyz, 2024-04-21:
So it should actually make the feature where the tip location is visible (for body containers). Otherwise it works.

closed 2025-03-30 by github-actions[bot]