FreeCAD/FreeCAD #5965: CarbonCopy should use names of constraints if available.

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4053

* **Reporter:** chrisb
* **Date submitted:** 7/12/2019
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** 

# Original report text

Instead of referencing dimensions by their constraint number, CarbonCopy should use the names where the user provided them.
[See forum discussion](https://forum.freecadweb.org/viewtopic.php?f=8&t=37638)

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

adrianinsaval, 2022-11-02:
here's the problem: 
https://github.com/FreeCAD/FreeCAD/blob/ec3a1838ca69e9a18216521961e766ce75fc2e1e/src/Mod/Sketcher/App/SketchObject.cpp#L6167-L6174

it is using the constraint index identifier, we need it to use the constraint name instead @abdullahtahiriyo do you know how to get the constraint name from the index? Or how can I get a pointer to the constraint so I can use
https://github.com/FreeCAD/FreeCAD/blob/da9ebc572f85f1694f75380c45ca2ac87f12f414/src/Mod/Sketcher/App/PropertyConstraintList.cpp#L86
assuming it does what I think it does: return the expression to access that constraint by checking if the constraint has a name and then if the name is valid for an expression return that or if it's not valid or it has no name return the index based expression.

adrianinsaval, 2022-11-02:
> do you know how to get the constraint name from the index?

would this be correct?
``` cpp
Constraint* scnstr =  psObj->Constraints[sourceid];
```
I found this note:
https://github.com/FreeCAD/FreeCAD/blob/da9ebc572f85f1694f75380c45ca2ac87f12f414/src/Mod/Sketcher/App/PropertyConstraintList.h#L97-L99
do I need to perform that check here? How do I do that?

adrianinsaval, 2022-11-03:
I tired this:
```cpp
                App::ObjectIdentifier spath;
                std::string scname = psObj->Constraints[sourceid]->Name;
                if ( App::ExpressionParser::isTokenAnIndentifier(scname) ) {
                    spath = App::ObjectIdentifier(psObj->Constraints) << App::ObjectIdentifier::SimpleComponent(scname);
                }
                else {
                    spath = App::ObjectIdentifier(psObj->Constraints) << App::ObjectIdentifier::ArrayComponent(sourceid);
                }

                std::shared_ptr<App::Expression> expr(App::Expression::parse(this, spath.getDocumentObjectName().getString() +std::string(1,'.') + spath.toString()));
                setExpression(Constraints.createPath(nextcid), expr);
```

but when the constraint is named `spath.toString` returns the path with a leading `.` while the index based one does not return a leading `.`
anybody knows what's up with this inconsistency? @wwmayer ?

wwmayer, 2022-11-08:
> do I need to perform that check here? How do I do that?

Actually you should always do it. 
```
Constraint* scnstr =  psObj->[sourceid];
if (scnstr) {
...
}
```

> anybody knows what's up with this inconsistency?

You can have a look at ObjectIdentifier::toString() but honestly I don't understand the logic there. The code is too complicated and I am not sure if it still exactly does what the documentation says.

adrianinsaval, 2022-11-08:
I looked but didn't understand it either :(
I had made a PR for this actually if you could have a look #7688 I just constructed the expression differently for each case but ideally `toString` should give consistent output.
> Actually you should always do it.

everything is happening inside this loop+conditional:
https://github.com/FreeCAD/FreeCAD/blob/d767a923a39d4a9bd8a0e248981e8b83fe12984f/src/Mod/Sketcher/App/SketchObject.cpp#L6154-L6163
I was thinking that maybe that already implies it is not null?

adrianinsaval, 2022-11-23:
my PR #7688 addresses this but additional changes will be necessary to support currently considered invalid names if/when 3062 is merged, with that PR names like `with spaces` or `1st` can be referenced with the syntax `Constraints[<<with spaces>>]` and `Constraints[<<1st>>]`

adrianinsaval, 2022-11-23:
@realthunder I think you did some work with `ObjectIdentifier`, if you are not too busy do you know why the inconsistent output of `toString` (see details above)? Is it a bug?

realthunder, 2022-11-24:
The leading dot is to indicate that the identifier is referencing a local property of the owner object (each expression must be owned by the object whose setExpression() is called). Without this leading dot, a current working expression may be broken in the future when you add an object with the same name as the property.

For example, `Constraint.Length` could mean an attribute named `Length` of the current object's property `Constraint`. Or, it could mean The property `Length` of an object named `Constraint`.

The leading dot is not needed for the indexer access because there is no ambiguity here. `Constraint[1]` always means to reference a local property `Constraint`.

adrianinsaval, 2022-11-24:
Ok but the leading dot would still be correct for the indexer right? wouldn't it be better to have consistent output and add a dot there too? Otherwise one has to treat the output differently for each case as I had to do in my PR #7688

commit c10330f7 (2022-12-02): [fixes #5965] CarbonCopy: if existing use constraint name in expressions
closed 2022-12-02 by 0penBrain