FreeCAD/FreeCAD #5966: FEM - Export transient FEM results to vtk/vtu for ParaView

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4060

* **Reporter:** Kunda1
* **Date submitted:** 7/17/2019
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

User Luftschraube created a macro and announced it in https://forum.freecadweb.org/viewtopic.php?f=22&t=37650 

[quote=Luftschraube]To give a little back to the community, I would like to share my macro with you: I used to perform quite a lot FEM analyses with FreeCAD, however, I prefer to export the results to VTK and then to visualise them using ParaView, which has a lot more options for that. Unfortunately, FreeCAD can only export one VTK file at a time, so I came up with a little macro that I use for transient analyses. The macro exports the FEM results and also writes a VTU file in the same folder. The VTU file can be simply opened in ParaView, and that way you directly have the transient results with the correct timestamps.[/quote]

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

luzpaz, 2022-12-03:
I've pasted the attached macro here for convenience:  
```python
#!/usr/bin/python
# -*- coding: utf-8 -*-
import FreeCAD
import Fem
import feminout.importVTKResults
import os
from sys import exit

FreeCAD.Console.PrintMessage(">>>---------Macro started\n")

#Get the file path and name of the active document 
p = FreeCAD.ActiveDocument.FileName
FilePath = "/".join(p.split("/")[:-1])
FileName = p.split("/")[-1].split(".")[0]
FreeCAD.Console.PrintMessage("File path:" + FilePath + "\n")
FreeCAD.Console.PrintMessage("File name:" + FileName + "\n")
FreeCAD.Console.PrintMessage("\n<<<----------Macro finished with errors\n")

#Get FEM result objects into a list
FreeCAD.Console.PrintMessage("Looking for result objects...")
FEMResultObjects = []

for obj in App.ActiveDocument.Objects:
	if obj.TypeId == "Fem::FemResultObjectPython":	
		FEMResultObjects.append(obj.Label)
		#FreeCAD.Console.PrintMessage(obj.Label + '\n')

NumberOfResultObjects = len(FEMResultObjects)
FreeCAD.Console.PrintMessage(str(NumberOfResultObjects) + " found\n")

if NumberOfResultObjects > 0:
	
	#Export result objects as .vtu file and generate .pvu file
	ExportFolderName = "vtk-export"
	ExportOutputPath = FilePath + "/" + ExportFolderName
	if not os.path.exists(ExportOutputPath):
		os.makedirs(ExportOutputPath)
		FreeCAD.Console.PrintMessage(ExportOutputPath + " has been created\n")
	else:
		FreeCAD.Console.PrintMessage(ExportOutputPath + " already exists\n")
	
	PVUFile = open(ExportOutputPath + "/" + FileName + ".pvd","w")
	PVUFile.write("<VTKFile type=\"Collection\" version=\"1.0\" byte_order=\"LittleEndian\" header_type=\"UInt64\">\n\t<Collection>\n")
	
	n = 0
	for ExportObject in sorted(FEMResultObjects):
		n = n + 1
		
		#Read and format time stamp (e.g., 12.37 s = 0001237)
		FreeCADResultLabelParts = ExportObject.split("_")
		PreDecimal = FreeCADResultLabelParts[-3]
		PostDecimal = FreeCADResultLabelParts[-2]

		if len(PostDecimal) == 1:
			PostDecimal = PostDecimal + str(0)
		TimeValue = float(PreDecimal) + float(PostDecimal)/100
		TimeString = str(int(round(TimeValue*100,0))).zfill(7)
	
		#For some reason, the VTK exporter expects a list (although only one object is processed at once...) - so build a list with the current object
		ExportObjectList = []
		ExportObjectList.append(FreeCAD.ActiveDocument.getObject(ExportObject))
		FileNameOUT = FileName + "_" + TimeString + ".vtu"
		FilePath = ExportOutputPath + "/" + FileNameOUT
		FreeCAD.Console.PrintMessage(FilePath + " is to be created\n")
		feminout.importVTKResults.export(ExportObjectList,FilePath)
	
		PVUFile.write("\t\t<DataSet timestep=\"" + str(TimeValue) + "\" file=\"" + FileNameOUT + "\"/>\n")
	
	#Close the .pvu file
	PVUFile.write("\t</Collection>\n</VTKFile>")
	PVUFile.close()
else:
	FreeCAD.Console.PrintMessage("Nothing to export!")

FreeCAD.Console.PrintMessage('\n<<<----------Macro finished\n')
```

tummechatronics, 2023-09-14:
@luzpaz  is this issue still relevant? If so I could you with it :)

luzpaz, 2023-09-15:
@FEA-eng I can't test this ATM. Can you weigh-in ?

FEA-eng, 2023-09-15:
> @FEA-eng I can't test this ATM. Can you weigh-in ?

I checked the attached macro in 0.21.1. It still works even though there were changes in results handling in this release.

luzpaz, 2023-09-15:
@tummechatronics  Want to tackle this ?

tummechatronics, 2023-09-15:
@luzpaz Yes, I would like to do it.

tummechatronics, 2023-09-17:
Hi @luzpaz, thanks for the assign. I have some questions:
I saw that @FEA-eng already tested in V0.21.1, and I recently did the same. Could you please give me some information on how to procedure with this issue?

luzpaz, 2023-09-17:
@timthelion can you post your coding questions on https://forum.freecadweb.org/viewtopic.php?f=22&t=37650

maxwxyz, 2024-09-21:
Any news on this?

ickby, 2025-03-23:
Exporting transients to paraview works now after merging #19610 

When a pipeline with multiple frames is exported, a *.vtm file is created, as well as a folder with all individual frames. This is default vtk behaviour. Now in Paraview one has two options:
1. Open the *.vtm file: This imports the multiblock file with each frame as a block. Not detected as time series within Paraview.
2. Go into the folder with the individual files. Paraview autodetects that they belong together, and when importing automatically creates a time series out of them. It even reads the correct values from for the timesteps.

Caveat: This export only works with the pipeline selected, not with the individual filter.

luzpaz, 2025-03-23:
@ickby thank you!  
Closing as fixed in #19610

closed 2025-03-23 by luzpaz