FreeCAD/FreeCAD #5968: Allow to select datums as direction for acting forces

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4064

* **Reporter:** uwestoehr
* **Date submitted:** 7/18/2019
* **FreeCAD version:** 0.18
* **Category:** Feature
* **Status:** new
* **Tags:** constraints

# Original report text

When setting the direction of a force it is not possible to select in the force setting one of the main axes or a datum line.

This seems to be only an UI issue.

This feature would be very helpful because no matter how you need to change your model later on, the main axes and datums remain as they are.

# Additional information

Forum thread: https://forum.freecadweb.org/viewtopic.php?f=18&t=37816

OS: Windows 7
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.17331 (Git)
Build type: Release
Branch: master
Hash: cf18161dc26267c259ac21675cb9637263bf4152
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: German/Germany (de_DE)

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

marioalexis84, 2024-03-02:
@FEA-english for me it can be closed as it is already implemented. Can you confirm?
It is now possible to select PartDesign datum, origin planes and origin axes objects.

maxwxyz, 2024-03-02:
@FEA-eng I'll close this as implemented. I can select origin planes and axes for force direction in FEM.

closed 2024-03-02 by maxwxyz

FEA-eng, 2024-03-02:
Right, I also checked this and it's already implemented. Good catch.

> @FEA-english

Wrong name in this mention ;-)