FreeCAD/FreeCAD #5971: [MacOSX Catalina] Unable to save (most likely due to security permissions)

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4073

* **Reporter:** Orcaman42
* **Date submitted:** 7/30/2019
* **FreeCAD version:** 0.18
* **Category:** Bug
* **Status:** feedback
* **Tags:** macOS

# Original report text

If you close the start page, then try to save a file, No save as window opens. Looking at thee file history there's nothing there that the file even previously existed either.

I don't think asking for help will work since users don't really safe files outside of using command S or the File menu. 

# Additional information

OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16131 (Git)
Build type: Release
Branch: (HEAD detached at 0.18.3)
Hash: 3129ae4296e40ed20e7b3d460b86e6969acbe1c3
Python version: 3.6.7
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

# Steps to reproduce

Make a new file, draw something. Close the start window & save your document. No Save As window opens, no save happens. Search for a file "Untitled.FCStd" & nothing.

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **OS: 10.15 10.15**
* **Platform:** MacOSX
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2019-07-30 12:51
Orcaman42 cant reproduce. How do you draw something without changing to another workbench (I'm still in Start WB). Also, next time please open a forum thread before opening a ticket (per our guidelines in the large yellow banner at the top of the page)

OS: Manjaro Linux
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16131 (Git)
Build type: Release
Branch: makepkg
Hash: 3129ae4296e40ed20e7b3d460b86e6969acbe1c3
Python version: 3.7.3
Qt version: 5.13.0
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/United States (en_US)

---

### Comment by JuM1697 2019-08-08 19:44
Hi! 
I can confirm that issue. And I might know the reason: With MacOS Catalina Apple asks you to give every Application permissions to access your folders to read/write files. Usually a prompt opens and asks for adjustment of the security settings when the application is trying to access files for the first time. This prompt doesn't appear!

OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16131 (Git)
Build type: Release
Branch: (HEAD detached at 0.18.3)
Hash: 3129ae4296e40ed20e7b3d460b86e6969acbe1c3
Python version: 3.6.7
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: German/Germany (de_DE)

---

### Comment by Kunda1 2019-08-12 17:06
JuM1697 please create a forum thread in the Help subforum referencing this ticket for discussion.

---

### Comment by JuM1697 2019-08-12 18:50
Okay, done!

---

### Comment by chrisb 2019-08-12 21:03
This is the [forum discussion thread](https://forum.freecadweb.org/viewtopic.php?f=3&t=38504).

---

### Comment by Kunda1 2019-09-24 09:27
another thread sprung up: https://forum.freecadweb.org/viewtopic.php?f=4&t=39587

luzpaz, 2022-06-21:
Related to https://github.com/FreeCAD/FPA/issues/6

luzpaz, 2023-08-06:
Closing due to macos bundles are signed and notarized (https://github.com/FreeCAD/FreeCAD/issues/9964#issuecomment-1664930953)

closed 2023-08-06 by luzpaz