FreeCAD/FreeCAD #5974: placement within a body breaks after a clone from non origin position

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4084

* **Reporter:** Jee-Bee
* **Date submitted:** 8/8/2019
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** new
* **Tags:** Clone, Part Design

# Original report text

The Clone of the PartDesign Body don't give the result as may expected from what the Clone should do(In my opinion). 
However my proposal could be broader interpreted as PartDesign... my issue is with PartDesign.

What is the situation:
Outside PartDesign all objects have a relation with Origin(general one).
In PartDesign normally you start with a Part OR Body. The Part have a relation with the PartOrigin. The Body have relation with PartOrigin OR Origin. The PartDesign Shape have a relation with BodyOrigin OR PartOrigin?
From now on i describe everything with the PartDesign Body as i discovered it with that. I assume the issue is the same with PartDesign Part.
When i make a Shape inside my Body the Shape have a relation with the BodyOrigin. When i change the Placement of the Body (moving), the relation between the Object and the BodyOrigin are constant(don't change). This is off course what i want!
Now a second instance of the same Body is needed. I can create a new Body and go trough all the steps as did before and give it a new Placement. This is Error prone so I create a Clone.
Now comes the Bug in my opinion. The Clone create a new Body as expected but the relation between the Object and the BodyOrigin aren't preserved!
In the Clone the Shape have exactly the same Position as the original Shape. The BodyOrigin is at the general Origin and the Placement is [0,0,0] (for the example i ignored Rotation!).

According to Crisb this is inherent the way FC handles Boolean AND Clone Objects.
In PartWB i understand that. with every Boolean function deciding what origin is choosed(master+slave/calculation/...) Yes i understand(a bit).
But PartDesign is another workflow. All Shapes have a relation to their origin from where they are created... So when Cloneing the Body means that the Shape and his relation to his BodyOrigin are preserved!

My proposal is to modify the PartDesign Clone so that Shape + BodyOrigin (+ Placement) are Preserved.

# Additional information

Link to Topic: https://forum.freecadweb.org/viewtopic.php?f=8&t=38295

# Steps to reproduce

- start new document
- (go to part design)
- create body
- create object in body (the object has a position/ relation regarding the coordinate system of the body)
- move body around
- create clone from body
internal coordinate system of the body from clone is different than the internal coordinate system of the original body. 
The clone internal coordinate system

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 2/15/2021

adrianinsaval, 2022-03-13:
This is very poorly formulated FR. Anyways, the placement of the clone object is actually initially the same of the original body and it can be set to 0 or whatever else desired. The placement of the body containing the clone is 0 but that is something different. IMO then the desired feature is already existing. I would close this issue.

Jee-Bee, 2022-08-03:
The whole bug is about the difference in placement between the original object and the newly created clone so that can't be a reason to close it
I just tested it. It is exactly as it was so it is not fixed and thus also no reason to close it:
[image]

adrianinsaval, 2022-08-03:
it is not a bug, it is intended behavior, use basefeature for what you want.
[image]

[basefeature.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/9252263/basefeature.FCStd.zip)

adrianinsaval, 2023-04-14:
As I said earlier this is not a bug per se since it's intended behavior, but it is indeed very unintuitive to manage PD clones and base features. It might be a good idea to eventually overhaul how these work.

Nekkowe, 2023-12-05:
Oh gosh, I just ran into this issue as a first-timer. What in the world. 
The origin ended up so far away from the actual part... moving it into place becomes a huge pain. 
Is there even any way to reset this to normal?
[image]

closed 2025-06-08 by github-actions[bot]