FreeCAD/FreeCAD #5975: Provide direct access to task panel dialog from Python

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4090

* **Reporter:** graffy
* **Date submitted:** 8/12/2019
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** GUI

# Original report text

In Python, there's no way to provide direct access to the task panel.  Typically, this can be managed easily enough by simply passing a reference to the form from the module that creates it to subsequent modules that may need it.

However, the C++ API already provides this accessibility via Gui.Control().activeDialog().  Unfortunately, the Python wrapper implementation merely returns a boolean indicating whether or not the active dialog exists.

Suggest rewriting the Python Control.activeDialog() to return a reference to the active dialog (or None).  This should pose little issue for existing code using Control.activeDialog() as any boolean test will still function properly.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by graffy 2019-08-13 13:01
Link to forum thread:

[https://forum.freecadweb.org/viewtopic.php?p=326529#p326529](https://forum.freecadweb.org/viewtopic.php?p=326529#p326529)

---

### Comment by rockn 2020-11-09 16:32
+1 for this feature request

I agree this would be best to have this implemented in 
`FreeCADGui.Control.activeDialog()` but maybe we can add this method instead : `FreeCADGui.Control.getActiveDialog()` to provide name and other interesting informations about the current activeDialog.

luzpaz, 2022-08-28:
@marioalexis84 would you consider giving this task a look at ?

wwmayer, 2024-11-09:
See https://github.com/FreeCAD/FreeCAD/commit/ff6f04dcc2493b2f8dae400e149a1a2cfe5f8d49

luzpaz, 2024-11-09:
Thanks wwmayer!  
Closing

closed 2024-11-09 by luzpaz