FreeCAD/FreeCAD #5977: Incorrect behavior of the Label tool in the Draft workspace

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4094

* **Reporter:** doubters
* **Date submitted:** 8/21/2019
* **FreeCAD version:** 
* **Category:** General
* **Status:** confirmed
* **Tags:** #post-to-forum

# Original report text

1. The label parameter "Display Mode" is not saved and, after opening the file, returns to "3D text"
2. Extra characters for Custom text on the working field (fixed in version 0.19)
3. When changing the "Display Mode" from "3D text" to "2D text", "Text Alignment" has to be changed several times to correctly position the inscription.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** General
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2019-08-21 20:03
You didn't add you full FC version (per the guidelines in large yellow banner top of page). Also, please open forum thread to signal to community about this issue. 

---

### Comment by doubters 2019-08-22 06:05
In which section of the forum to post?

Information about versions of FreeCADs that were checked:

OS: Ubuntu 19.04
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.3.
Build type: Release
Python version: 3.7.3
Qt version: 5.12.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: C/Default (C)

OS: Ubuntu 19.04 (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.
Build type: Release
Python version: 3.7.3
Qt version: 5.12.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: C/Default (C)

---

### Comment by Kunda1 2019-08-22 15:28
doubters https://forum.freecadweb.org/viewforum.php?f=3

---

### Comment by Kunda1 2020-01-09 19:08
carlopav care to test this to see if in fact it's an issue ?

---

### Comment by carlopav 2020-01-09 19:20
kunda1, i can confirm the behaviour with:
```OS: Windows 7 SP 1 (6.1)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.19093 (Git)
Build type: Release
Branch: master
Hash: 9e4710d679d44613a1ae4264b86ef9f6d12f4e5a
Python version: 3.6.8
Qt version: 5.12.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: Italian/Italy (it_IT)```

I never had a serious look at this part of Draft code, so I don't know what's happening, but anyway this is not an intuitive behaviour. When I'll have some time i could give it a deeper look, but not so soon...

---

### Comment by Kunda1 2020-01-10 20:45
Thanks carlopav! I'm marking this ticket 'confirmed' and setting the target for 0.19
If you do have time that is great but no worries.

luzpaz, 2022-04-12:
@Roy-043 I can't make heads or tails of this ticket. And the clip in the original is lost. Any idea?

Roy-043, 2022-04-12:
> 1. The label parameter "Display Mode" is not saved and, after opening the file, returns to "3D text"

This issue does not occur in V0.20, so has presumably been fixed in the meantime.

> 2. Extra characters for Custom text on the working field (fixed in version 0.19)

There was an issue where removing lines from a multi-line label did not work properly. I suppose the OP is referring to that, but I can't be sure. The OP says that it has been fixed anyway.

> 3. When changing the "Display Mode" from "3D text" to "2D text", "Text Alignment" has to be changed several times to correctly position the inscription.

In the current version `TextAlignment` is ignored if the `DisplayMode` is `"2D text"`.
See https://github.com/FreeCAD/FreeCAD/pull/5266

This issue can be closed.