FreeCAD/FreeCAD #5978: Material allowance does not seem to differentiate between air or unused stock and remaining parts of the model.

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4096

* **Reporter:** freman
* **Date submitted:** 8/23/2019
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

Face milling path tool does not clean the face, it leaves residual bits in the corners. Using "material allowance" can extend this but it will eat into other parts of the model if it is not simply the top surface of the workpiece. 

https://forum.freecadweb.org/viewtopic.php?f=15&t=38609

I see two principal issue here:
1. Material allowance does not seem to differentiate between air or unused stock and remaining parts of the model, meaning it will happily remove part of the work piece. I would consider that a serious bug.
2. A facing operation by default does not clear the selected face and thus does not fulfil its primary task in any situation. It requires a specific option parameter to be set by the user to clear the corners even if there is nothing but air/stock outside the boundary. Once the above issue is resolved , this could probably be handled in a more helpful way, overstepping by the tool radius where it does not damage the surrounding part. Material allowance could then be used to set more over step if desired to get a cleaner finish ( no turns on the work ).

If there is a step in the model, this requires some rather imaginative and ugly workarounds to clear a face, while there are obvious tools paths to achieve the result.
 
https://forum.freecadweb.org/download/file.php?id=90043

# Additional information

OS: Linux (LXDE/LXDE)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.16988 (Git)
Build type: Unknown
Branch: master
Hash: ff7975291d33a88e6a8282b88a62dbcbcc01ba2b
Python version: 2.7.15
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedKingdom (en_GB)

# Steps to reproduce

Open the linked FreeCAD model and look at the face mill operations. Run the sim. and note that path never goes beyond the boundary of the face. Add a "material allowance" of half tool radius to clean up and note that it takes a chunk out of the central block of the work.

PS. Ratonlaveur notes that this is not what "material allowance" usually refers in conventional machining language. This should probably be changed so that FreeCAD is not defining a new use of standard terms. 

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **Updated:** 2/6/2021

sliptonic, 2022-04-06:
Is this even an issue?  Which part of this is not resolved by extensions?
Will close if no reason to keep open.

Russ4262, 2022-04-07:
0.20 pre-release analysis with Revision 28638:

My thinking at the moment as it pertains to preparations for 0.20 release is similar to that of @sliptonic, as posted above.  I am convinced that the Pocket op with the Extensions feature should allow for many use cases where MillFace is currently failing due to the bug reported here.  In my opinion, priority is low for this bug, while per MillFace intended behavior, I suggest this bug stay active as a reminder that collision avoidance needs to be addressed, and by this issue report, it is acknowledged.

The bug reported here pertaining to collision with the model exists in other ops, including Pocket, that also use a simple direct offset adjustment of the face region, which is then processed for path generation.  This is known by devs, and will be easier to address systematically as the refactoring of Path modules continues.

closed 2023-08-12 by sliptonic