FreeCAD/FreeCAD #5984: Improve tooltips for the FC preference panel (make them more verbose) using the content from the wiki

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4120

* **Reporter:** Kunda1
* **Date submitted:** 9/9/2019
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** #ready-for-migration, tooltips

# Original report text

Relevant wiki page: https://www.freecadweb.org/wiki/Preferences_Editor
source code: /src/Gui/Dlg*.ui

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 1/25/2022

donovaly, 2022-04-13:
Unless there will be a PR the next days, I would like to remove this issue from the 0.20 milestone.

sliptonic, 2023-04-13:
This issue appears to be two separate things.  One is about making the tool tips more useful.  There is an existing issue for [informative tool tips](https://github.com/FreeCAD/FreeCAD/issues/9253).  

The other issue is about the quality of the wiki page.  I agree with @PaddleStroke that the wiki page needs rework. That is an issue for the wiki. 

I suggest that this issue be re-purposed to be exclusively about the wiki page.

sliptonic, 2023-04-13:
@luzpaz Can you change the title to be more reflective of the goal

sliptonic, 2023-04-13:
> Is it still worth keeping this open since it goes without saying that the wiki needs to be updated and the tooltip issue has been superceded by #9253 ?

If we close it without doing the wiki change, the required change may never happen.  The issue needs to get more specific about exactly what should be done with the wiki

Roy-043, 2023-05-04:
Good grief, this is beyond annoying. Nobody has taken the effort to actually LOOK at the Wiki page. Please pay particular attention to the word 'Expand'. Also note that the page has been updated recently.

sliptonic, 2023-05-04:
> Good grief, this is beyond annoying. Nobody has taken the effort to actually LOOK at the Wiki page. Please pay particular attention to the word 'Expand'. Also note that the page has been updated recently.

If the issue is resolved, close it

closed 2023-05-04 by sliptonic