FreeCAD/FreeCAD #5985: HLR Returns No Edges for Shape

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4122

* **Reporter:** wandererfan
* **Date submitted:** 9/11/2019
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** acknowledged
* **Tags:** OCC 7.3, upstream

# Original report text

[Forum discussion](https://www.forum.freecadweb.org/viewtopic.php?f=3&t=39216)

OCC HLR (HLRBRep_Algo) returns empty result for attached shape.  DRAWEXE fails to display result of HLR with message "Error: AIS_Shape::Compute() HLR Algorithm has failed ()".

# Additional information

DRAWEXE fails to display shape in "algo" mode, but displays shape in "polyalgo" mode.

# Steps to reproduce

Open attached FCStd file and create new View for object "Bullet".   Empty view will be created with no error messages.

Changing View to Coarse mode produces result. 

Note: FC in OSX will display bullet shape, but Windows or Linux.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Platform:** win, linux, not OSX
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by wandererfan 2019-09-11 15:21
[OCC bug report](https://tracker.dev.opencascade.org/view.php?id=30961)

---

### Comment by wandererfan 2020-06-21 13:01
Not fixable in FreeCAD

luzpaz, 2023-04-15:
@WandererFan do you mind bumping the upstream ticket?

grubertechnic, 2023-05-15:
@WandererFan : see https://forum.freecad.org/viewtopic.php?p=675975#p675975 , closing?

WandererFan, 2023-05-15:
I bumped the OCC ticket.  There's nothing to be done on our side. 

@luzpaz - do we leave tickets like this open?

grubertechnic, 2023-05-15:
@WandererFan : the only issues are here
 - the shape is invalid, the curvature of the spline in undefined since the tangency point is the same as the start point (curvature is infinity/0)
 - that the OCCT geometry check does not work
 
But when you repair the tangency points it works flawless. I do not see an issue here regarding FreeCAD or OCCT (for the geometry check, i would make a separate ticket, since there are more checks they regularly incorrect).

grubertechnic, 2023-06-30:
@WandererFan : here a video, that shows that it is a user error, the tangency knot overlaps.

[2023-06-30 22-52.webm](https://github.com/FreeCAD/FreeCAD/assets/109620914/c3e084e1-2e76-4919-9ffe-412312d14321)

file:
[Temp_cb.zip](https://github.com/FreeCAD/FreeCAD/files/11923199/Temp_cb.zip)

imho: should be closed

closed 2023-07-14 by WandererFan