FreeCAD/FreeCAD #5987: 0 length lines in sketch introduce bugs at Part-> Revolve time

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4126

* **Reporter:** baltic
* **Date submitted:** 9/13/2019
* **FreeCAD version:** 0.18
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

Trying to **Part->Revolve** a sketch, which contains 0 length lines produces wrong results. 
And **Validate Sketch** tool does not catch 0 length lines.

# Additional information

OS: Ubuntu 18.04.3 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16131 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.18.3)
Hash: 3129ae4296e40ed20e7b3d460b86e6969acbe1c3
Python version: 3.6.7
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

[forum link](https://forum.freecadweb.org/viewtopic.php?f=3&t=39274)

# Steps to reproduce

Create a simple sketch in XY plane like that:

[img here...](https://freecadweb.org/tracker/file_download.php?file_id=2338&type=bug)

On the top part of it add 2 fillets:

[img here...](https://freecadweb.org/tracker/file_download.php?file_id=2339&type=bug)

Constrain their radiuses to 1mm

[img here...](https://freecadweb.org/tracker/file_download.php?file_id=2340&type=bug)

Constrain the thickness of the wall to 2mm

[img here...](https://freecadweb.org/tracker/file_download.php?file_id=2341&type=bug)

Now you effectively have 0 length line between the 2 arcs.
When you rotate it through **Part->Revolve** tool around Y axis, you get a non hollow object:

[img here...](https://freecadweb.org/tracker/file_download.php?file_id=2342&type=bug)

But it supposed to be hollow. When you delete the 0 length line from sketch, the tool works correct.

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **OS: Ubuntu 18.04**
* **Platform:** 64
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by openBrain 2019-09-13 17:21
According forum, community proposal is to add a tool in ValidateSketch to find/fix this 0-length lines.

---

### Comment by baltic 2019-09-14 18:07
Original ones, instead of scaled down copies

grubertechnic, 2023-07-08:
as @PLChris (chrisb) pointed out in the forum, it is only an user error (no coincident lines endings, small lines, 0mm constraints), should be closed.

closed 2023-07-09 by luzpaz

grubertechnic, 2023-07-12:
@luzpaz : thanks