FreeCAD/FreeCAD #5989: Crash on setting certain dimensions in a sketch (Upstream OCC bug)

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4129

* **Reporter:** uwestoehr
* **Date submitted:** 9/13/2019
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** acknowledged
* **Tags:** #ready-for-migration, OCC Bug, OCC Unscheduled, sketcher, upstream

# Original report text

- open the attached example file
- open the Sketch "Sketch-Pinpad"
- set for the Edge 2 a vertical constraint (e.g. 3mm)
- close the sketch

result: crash
One out of 5 times it does not crash and I get an "access violation" in the report view.

The crash could be reproduced also on Mac.

TheMarkster analyzed the problem in this post:
https://forum.freecadweb.org/viewtopic.php?p=327107#p327139

# Additional information

Forum thread: https://forum.freecadweb.org/viewtopic.php?f=3&t=38590

OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.17505 (Git)
Build type: Release
Branch: master
Hash: 755536e9df94d2d39da1468420f1fd333c35da7a
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: German/Germany (de_DE)

# Other bug information

* **Priority:** normal
* **Severity:** crash
* **Category:** Bug
* **Updated:** 1/24/2022

# Discussion from Mantis ticket

### Comment by abdullah 2020-05-29 13:51
Note: OCCT crashes in PartDesign::FeatureFillet

---

### Comment by abdullah 2020-05-30 14:00
I have reported the bug upstream to OpenCascade and provided a patch for a possible solution. With the patch an exception is raised, which is catched by FreeCAD and the crash is avoided:
https://tracker.dev.opencascade.org/view.php?id=31585

kunda1
I think you do something with upstream reported OCCT tickets... so I let you do your magic ;)

---

### Comment by Kunda1 2020-05-30 15:15
abdullah I've added this ticket and companion upstream tickets to the thread tracking upstream OCC bugs: https://forum.freecadweb.org/viewtopic.php?t=20264 ;)

---

### Comment by abdullah 2020-06-06 17:45
There is no solution other than wait for OCCT to solve the access violation:
https://tracker.dev.opencascade.org/view.php?id=31585

Until them, this ticket remains as acknowledged.

---

### Comment by Kunda1 2020-12-01 11:39
Added 'OCC bug' to the bug summary for easier Identification

---

### Comment by Kunda1 2022-01-24 18:40
abdullah do you mind bumping the upstream ticket?

adrianinsaval, 2022-02-12:
@chennes similar situation as before, OCC bug reported by @abdullahtahiriyo with a possible fix but no response. In case you can do something :)

chennes, 2022-02-13:
It is the same fix, this is actually the same error as in #5604 . Patch is pending, I will update both tickets as it progresses through the OCCT process.

adrianinsaval, 2022-02-13:
Oh sorry I did not realize 😅

luzpaz, 2022-04-11:
@chennes is this good to go?

chennes, 2022-04-11:
The patch has been merged upstream.

luzpaz, 2022-04-11:
Closing ticket then.

closed 2022-04-11 by luzpaz

luzpaz, 2022-07-29:
Forgot to notate https://forum.freecadweb.org/viewtopic.php?f=10&t=20264  
Patch was merged in OCC 7.6.2  
(Late) congrats @abdullahtahiriyo :laughing: Thanks!