FreeCAD/FreeCAD #5991: Add PartDesign Common operations

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4135

* **Reporter:** townkat
* **Date submitted:** 9/17/2019
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** Part Design

# Original report text

Please add Common (intersection) operations in PartDesign in addition to Additive and Subtractive features.

More info in the following topics:
https://forum.freecadweb.org/viewtopic.php?f=8&t=35755
https://forum.freecadweb.org/viewtopic.php?f=8&t=39297

Thank you.

# Other bug information

* **Priority:** high
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

luzpaz, 2023-11-08:
@FlachyJoe is this a duplicate ?

FlachyJoe, 2023-11-08:
@luzpaz I don't know if it's shown elsewhere but I see SubstractiveHelix has an Outside property which do this. It seems quite easy to implement for the other subtractive features.

luzpaz, 2023-11-11:
> seems quite easy to implement

@FlachyJoe go for it ? or perhaps if you don't have time, can you elaborate here how a dev would go about adding it ?

FlachyJoe, 2023-11-11:
The implementation for SubstractiveHelix should be duplicate for all features (maybe a common util function can be write)
Ref: https://github.com/FreeCAD/FreeCAD/blob/66ece1c0684dd6b9757a2c39a5479713555f8897/src/Mod/PartDesign/App/FeatureHelix.cpp#L344-L356

@bgbsww are you interested?

bgbsww, 2023-11-11:
@FlachyJoe I'll take a look.

bgbsww, 2023-11-12:
TL;DR assign me please.

So, the App side of this is pretty trivial, as @FlachyJoe identified above.  The Gui side is not as straightforward, since there is no Giu/TaskAddSubParameters in the inheritance, which means either a bunch of ugly cut and paste code in each of the six subtractive types to be added, or adding to the inheritance tree and refactoring a bit.

Also, the Helix code seems to be clearly newer and substantially different than the older AddSub subtypes, which is another place to consider refactoring to consistency.

I have a proof of concept working just fine for pocket so I'm confident, there are just a lot of moving pieces here to make it clean.

If anyone sees something I'm missing, I'm all ears.

NomAnor, 2024-03-10:
I have an idea about refactoring the PD features, this would also allow this to be implemented: https://forum.freecad.org/viewtopic.php?t=85967

maxwxyz, 2024-08-30:
@bgbsww could you also look into https://github.com/FreeCAD/FreeCAD/issues/11209 for this fix https://github.com/FreeCAD/FreeCAD/pull/11580

maxwxyz, 2025-01-22:
@NomAnor are you still open for working on this?

NomAnor, 2025-03-09:
In principle yes. I could see this as part of a bigger refactoring of PD. Maybe even some unifaction with Part. But I lack the vision how to tackle it.

ant1ucem, 2025-04-06:
> But I lack the vision how to tackle it.

A very important feature indeed — it would save a lot of operations. I suggest implementing it in the following way:

1. Merge the pairs of buttons like Pad/Pocket, Revolution/Groove, Additive Pipe/Subtractive Pipe, and so on.
2. Add a toggle switch at the top of the task panel to select between Add, Subtract, and Common operations.

[image]

closed 2026-09-03 by kadet1090