FreeCAD/FreeCAD #5992: Start WB could report more details when highlighting an FCStd file

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4142

* **Reporter:** Kunda1
* **Date submitted:** 9/24/2019
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** Start Workbench

# Original report text

sliptonic makes a great point in his YT clip [1] about the StartWB adding the ability to report more information on a file when highlighting or hovering over it. 

[1] https://youtu.be/uKD6sGdci4Y?t=189

The whole segment starts at https://youtu.be/uKD6sGdci4Y?t=53
Another interesting part https://youtu.be/uKD6sGdci4Y?t=114

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

luzpaz, 2023-11-13:
@tmksw any interest looking in to this feature ?

tmksw, 2023-11-14:
@luzpaz  From the attached video I understood that it's about adding the full path of the drawing file to the thumbnail when hovering over recent file in `Start WB`. Is that correct, what exactly needs to be done in this specific task ?

luzpaz, 2023-11-14:
@sliptonic would you have a moment to weigh-in ?

yorikvanhavre, 2023-11-16:
We could add more information like:

* The full path of the file
* The author
* The license
* The number of objects maybe?

tmksw, 2023-11-19:
PR for above mentioned changes is ready.

*Before*

[image]

*After*

[image]

PR #11450 (2023-11-20): [Start WB] extend file info in MRU tooltip
Additional information has been added to MRU tooltip.

*Before*

[image]

*After*

[image]

This resolves #5992
closed 2023-11-20 by adrianinsaval