FreeCAD/FreeCAD #5995: Show spiral length in the property editor

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4159

* **Reporter:** vocx
* **Date submitted:** 10/9/2019
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

Part Helix and Part Spiral (which are derived from Helix) objects don't show a length property in the property editor.

The length is included in the Shape, so it just has to be displayed as a read-only value.
```
obj.Shape.Length
```

See this thread [Length of a spiral](https://forum.freecadweb.org/viewtopic.php?f=3&t=39884)

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

luzpaz, 2022-12-31:
CC @adrianinsaval @chennes @Roy-043 (participated in the forum thread)

closed 2023-11-27 by chennes

luzpaz, 2023-11-27:
Nice! Thank you @bgbsww !

PR #11406 (2023-11-27): Add readonly length properties to part helix and spiral
Fix #5995