FreeCAD/FreeCAD #6000: Part Design auto-generated Body not placed under existing Part

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4178

* **Reporter:** garya
* **Date submitted:** 10/27/2019
* **FreeCAD version:** 0.18
* **Category:** Bug
* **Status:** confirmed
* **Tags:** Part Design

# Original report text

A sketch which auto-creates a Body does not place it under the active Part.  Instead, it is placed at the same level as the active part.  It should be placed under the part.
See [forum topic](https://forum.freecadweb.org/viewtopic.php?f=3&t=40258&p=342235#p342235)

# Steps to reproduce

Create a new document
Part Design WB
Create Part
Create Sketch, select xy plane, click ok
Close the sketcher task
View the model tab
The auto-generated Body is at the same level as the Part; it should be under it.

# FreeCAD Info

```
OS: FreeBSD 11.2-RELEASE-p4
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.0.18.2
Build type: Release
Python version: 3.6.9
Qt version: 5.12.2
Coin version: 3.1.3
OCC version: 7.3.0
Locale: C/Default (C)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: FreeBSD 11.2-RELEASE-p4**
* **Platform:** PC
* **Updated:** 2/6/2021

luzpaz, 2022-12-03:
Is this by design, @wwmayer  ?

wwmayer, 2022-12-06:
> Is this by design

No, it's should be added to the Part container.

commit 3e33a91a (2022-12-07): PD: fixes #6000: auto-generated Body not placed under existing Part
closed 2022-12-07 by wwmayer