FreeCAD/FreeCAD #6003: Add an addition usage to a current icon in sketcher

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4184

* **Reporter:** freedman
* **Date submitted:** 11/4/2019
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

Many designers use Sketcher in Normal and 3D camera view. We have a current icon that when clicked will change the camera view to "normal to sketch". I would like an addition to the use of this icon, when clicked on it could toggle between sketch normal and the saved prior view before entering sketcher. Here is a link to a discussion, page 2 of the posts.
https://forum.freecadweb.org/viewtopic.php?f=8&t=40317&start=10

# Additional information

Below is an image of the icon for "change view perpendicular to sketch", it's in the Sketch toolbar list at the bottom. It is a sketch with a magnifying glass on it.
The version number and platform are unimportant for this discussion.

# FreeCAD Info

```
OS: Windows 7
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16085 (Git)
Build type: Release
Branch: master
Hash: 6323069cbbda5ac71dff073f15cc632d8a1f3ea3
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.2.0
Locale: English/UnitedStates (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Platform:** See below
* **Updated:** 2/6/2021

adrianinsaval, 2023-05-23:
Might be a better idea to offer a preference to keep the camera position when opening a sketch, or have a separate command

Roy-043, 2023-10-05:
Duplicate: #5819

closed 2023-10-05 by Roy-043