FreeCAD/FreeCAD #6004: Sketcher opens a sketch at the current location, we need an icon to take us to the sketch.

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4185

* **Reporter:** freedman
* **Date submitted:** 11/4/2019
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

Sketcher needs a way to center selected sketch elements in sketch window. My example in the thread; If you have a 10 story building and your looking at the first floor, you click to edit a sketch of something on the 10th floor, sketcher will open a window at the current view location, we need an icon to center the sketch elements in the sketch window, in this case it would chage the view to the 10th floor. Sketcher does not do this currently. Here is a link to a discussion, about page 3.
https://forum.freecadweb.org/viewtopic.php?f=8&t=40317&start=10

# Additional information

The new icon could be a sketch with a bulls-eye on it to depict a focus of the sketch elements.
Thank you 

# FreeCAD Info

```
OS: Windows 7
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16085 (Git)
Build type: Release
Branch: master
Hash: 6323069cbbda5ac71dff073f15cc632d8a1f3ea3
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.2.0
Locale: English/UnitedStates (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Platform:** See below
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by freedman 2019-11-07 01:33
I have started a new thread asking for more advice.
https://forum.freecadweb.org/viewtopic.php?f=8&t=40643

This thread addresses the issue directly. From what I see Mario52 has solved the problem with his code. Included in the thread is code to test. This seems like a huge improvement to sketcher. If this code is run when entering sketcher the target sketch is focused in the screen.
https://forum.freecadweb.org/viewtopic.php?f=22&t=40938#p348173

Zolko-123, 2022-10-25:
should be closed

luzpaz, 2022-10-26:
@Zolko-123 thanks for the feedback but would you mind elaborating as to why? (you've commented this way on some recent tickets already)

Zolko-123, 2022-10-26:
it's unclear what it wants and obviously not followed.

adrianinsaval, 2022-10-26:
I agree this should be closed, it's poorly formulated, to me it sounds like they want the fit all or fit selected functionality which already exists

closed 2022-10-26 by luzpaz