FreeCAD/FreeCAD #6007: Expression parser does not allow "/" after first member

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4196

* **Reporter:** babaroga
* **Date submitted:** 11/18/2019
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** confirmed
* **Tags:** expressions

# Original report text

If someone try to enter eg. 12/2 in any field for X, Y, Z, constrain in sketcher, parser will not allow to type "/" sign just after first member in expression. All other operations works as expected.
Everything works OK if entered as formula.

Issue is described in

[Forum thread](https://forum.freecadweb.org/viewtopic.php?f=3&t=40995)

# Steps to reproduce

Try to change X, Y or Z position of eg. body with

12/2 - does not work
(12)/2 - does not work
0+12/2 - works

# FreeCAD Info

```
OS: Manjaro Linux (KDE//usr/share/xsessions/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.18753 (Git)
Build type: Release
Branch: master
Hash: c0a80e1a9a5683404fe506201d0ee74b28ea819d
Python version: 3.7.4
Qt version: 5.13.2
Coin version: 4.0.0a
OCC version: 7.4.0
Locale: English/United States (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Manjaro 18.1.3**
* **Platform:** Linux
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by openBrain 2019-11-22 16:49
babaroga : thanks for reporting. However please notice that per our guidelines (big yellow banner) you should have open a topic in our forum before reporting.
Though your report is correctly filled in, bug can be confirmed and seems unknown up to now, we'll keep this ticket as is without further request to post to forum. ;)

---

### Comment by openBrain (private) 2019-11-22 16:50
Kunda1 : also this one to move to Expressions project ? :)

---

### Comment by babaroga 2019-11-22 16:55
Hello,

openBrain It was reported in forums on

[https://forum.freecadweb.org/viewtopic.php?f=3&t=40995](https://forum.freecadweb.org/viewtopic.php?f=3&t=40995)

Somehow, link is not visible in my first post.

---

### Comment by openBrain 2019-11-22 17:23
babaroga :there was no text in the url tag in the description. I fixed it. ;) 
Could you edit the title of the forum thread and prepend "[Bug 4196]" to it (thread title is actually OP title)? Thx

---

### Comment by Kunda1 2019-11-28 12:52
Moved to 'Expressions' category

---

### Comment by ulrich1a 2019-12-31 16:07
The reported bug is not caused by FreeCAD-Expressions. Expressions are used in the dialog reachable via the blue "fx"-symbol: !(https://www.freecadweb.org/wiki/File:Sketcher_Expressions.png). This dialog is not affected by the bug here and behaves different and uses different code. 
The bug is related to the input field of the Gui, which uses a different parser. The formula typed in the input field is gone after its calculation. 

Both usages uses "expressions" but are coded in different areas of FreeCAD.

Ulrich

---

### Comment by openBrain 2020-01-01 17:47
ulrich1a : would you be able to give a pointer to involved parser code in GitHub ?

luzpaz, 2023-02-22:
Cleaned up this ticket. 

@0penBrain question, should i remove the Expressions tag since per Ulrich this isn't related to the expressions engine ?

0penBrain, 2023-02-22:
@luzpaz indeed this is the formula parser involved here, not the expression one.
Also in author description, "formula" is used instead of "expression".

adrianinsaval, 2023-04-26:
can't reproduce, most likely fixed by #7124

closed 2023-04-26 by adrianinsaval