FreeCAD/FreeCAD #6011: BlendCurve - need it to accept ANY point / line as an argument

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4204

* **Reporter:** Pauvres_honteux
* **Date submitted:** 11/25/2019
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

See forum discussion: [3D parametric curve.](https://forum.freecadweb.org/viewtopic.php?f=3&t=38299)
Adding to the list: selecting points / lines from the tree as well from 3D view.

# FreeCAD Info

```
OS: openSUSE Leap 15.1 (KDE//usr/share/xsessions/default)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.Unknown
Build type: Release
Python version: 3.6.5
Qt version: 5.9.7
Coin version: 4.0.0a
OCC version: 7.4.0
```

# Other bug information

* **Priority:** high
* **Severity:** major
* **Category:** Feature
* **OS: All **
* **Platform:** All
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Pauvres_honteux 2019-11-25 10:44
Clarification: this request is of course directed to Part / Part Design. Not directly at Chris_G.

luzpaz, 2022-08-27:
@tomate44 do you mind weighing-in on this now that BlendCurve is partially implemented as of recently ? TIA!

tomate44, 2022-08-28:
I'll have a look.
Maybe you can assign this bug to me ?