FreeCAD/FreeCAD #6026: No "snap to intersection" implemented in Sketcher

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4253

* **Reporter:** Bernhard Schiffner
* **Date submitted:** 1/21/2020
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** #post-to-forum

# Original report text

I'am used to work a lot in construction mode (blue lines), and anchor object lines (white lines) to points defined there later in the process.
I noticed that there is no "snap to intersection" feature implemented in FreeCAD / Sketcher.
If compared to other CAD tools this is (for me) something essential missing.

The example attached, gives a nice opportunity to test the problem.

On the way you can notice, that the order of snapping is improveable too.
(Snap to the next construction point first and not the next line possible.)

To understand the example: it symbolizes the end of a shaft entering a gear. The sides of the triangle are milled a little bit "malicious" and don't fit the regular hexagon.
An (example) shape is given, but I'd like to do the equivalent using blue construction lines and their intersections for defining points.

# Additional information

You can use workarounds like defining a point by "snap to line" to two different lines.
But if you close the contour this way, sketcher starts to claim nonsense about redundant dependencies. :-)

(Using self compiled pyside2 packages to make FreeCAD run with KDE NEON.)

# Steps to reproduce

Try to catch two intersecting construction lines to define a point.

# FreeCAD Info

```
OS: KDE neon User Edition 5.17 (KDE//usr/share/xsessions/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.
Build type: Release
Python version: 3.6.9
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: German/Germany (de_DE)

```

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by openBrain 2020-01-21 16:23
Thx for suggesting. However, could you open a topic in our forum so the proposal is discussed by the community (you can also get there help about the 'redundant constraint' issue) ? Thx

---

### Comment by Bernhard Schiffner 2020-01-21 20:41
I'll start this discussion.
I am new at this, so please excuse possible errors from my side.
For the moment I add some (possible) icon for the idea.

Where to snap first ...
1.) Snap to point (removes 2 DOF by using 1 user created constituent)
2.) Snap to intersection (removes 2 DOF by using 2 user created constituents)
3.) Snap to line (removes 1 DOF by using 1 user created constituent)
....
x.) vertical constraint (removes 1 DOF by using 1 general constituent)

---

### Comment by Kunda1 2020-01-24 11:28
OP, please post to the forum per the guidelines (big yellow banner at the top of this page). Please do not forget to crosspost the URL of the thread back here.

---

### Comment by Kunda1 2020-06-07 14:33
Bernhard_Schiffner forum thread pls

---

### Comment by Bernhard Schiffner 2020-06-07 20:13
If this forum post of 2017 (as an example) is more or less about the same thing, is it necessary to open a new thread there?

https://forum.freecadweb.org/viewtopic.php?f=8&t=24815&p=194830&hilit=snap+intersection#p194830

adrianinsaval, 2023-04-15:
My interpretation of this is that it wants to add auto constraints (not really snaps) using two point on object on both edges when the mouse pointer is near their intersection. I think that could be a useful feature, thoughts @abdullahtahiriyo @PaddleStroke @AjinkyaDahale ?

PaddleStroke, 2023-04-15:
This would require to expand the snap functionality because currently it
uses the preselected element.
This would need to have pointOnRay to return all instead of only the first
element. It's related technically to the issue on pick radius / elements
hard to select.

abdullahtahiriyo, 2023-05-02:
I am not sure if this qualifies as "snap", if it does I have problems to see a clear advantage.

1) If I am creating a point, it will do an autoconstraint to only one. This is an advantage. It won't autoconstraint to an intersection. It may be possible to extend autoconstraints to detect intersections between curves and then apply two PoO constraints.

2) If I am dragging a point, then it will snap to object, but it won't autoconstraint. Then, we could think about extending snapping. However, is it useful in this case to snap to two objects without any constraint? I mean, any movement of one of the curves will lead to the point not being in the intersection anymore after snapping.