FreeCAD/FreeCAD #6029: Problem with negative offset "Placement"  when the preference are "Building US" units

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4263

* **Reporter:** jylandry
* **Date submitted:** 2/10/2020
* **FreeCAD version:** 0.18
* **Category:** Bug
* **Status:** new
* **Tags:** #pending-forum, #ready-for-migration, units

# Original report text

https://forum.freecadweb.org/viewtopic.php?f=8&t=42766

When using "Placement" for an object and working with the unit in "Building US(ft-in/sqft/cuft)"

You can't work with negative number.  For example, «-1.5"» or «-1+1/2"» will translate to «-1/2»

The same for «-13'6"» that will become «-12'6"»

# Steps to reproduce

1) Put your preference in "Building US(ft-in/sqft/cuft)"

2) Create an empty document with an object like a cube (or anything will do)

3) try to change the "placement"  (Translation) -- any of X: Y: or Z:    -- Enter «-1.5"» or «-1+1/2"» or «-13'6"»

if you return in placement, the entry «-1.5"» should have been change to «-1/2"»
the entry «-13'6"» shall have been change by «-12'6"»

# FreeCAD Info

```
OS: Ubuntu 18.04.4 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.4.
Build type: Release
Python version: 3.6.8
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 2/5/2022

# Discussion from Mantis ticket

### Comment by openBrain 2020-02-10 11:00
Could that be a duplicate of #3925 ?

---

### Comment by jylandry 2020-02-11 02:16
I look at it. Some aspect of  0003925  might exhibit the same symptom.

However, their is additional problem with the Building US unit related to the interpretation of the fraction.

a) When you enter -10+1/8" and click OK on the "Placement box"... you get -9+7/8". This might be similar to 0003925.

b) When you re-enter the "Placement Box", it does display  -8+1/8".  I don't think it is the same.

---

### Comment by jylandry 2020-02-11 02:17
if you look at the post	https://forum.freecadweb.org/viewtopic.php?f=8&t=42766

I have documented several cases scenario... including one that is not flagged as a bug.

---

### Comment by pablito 2020-09-12 15:07
Independently observed again in 0.19.22366 and discussed in thread : https://forum.freecadweb.org/viewtopic.php?p=429940#p429940.

---

### Comment by Kunda1 2022-01-23 00:02
openBrain since you've been tackling 'units' do you want to assign yourself to these tickets tagged as 'units' ?

---

### Comment by openBrain 2022-01-25 15:34
Kunda1 : certainly not. This ticket implies dealing with lexers and parsers, that I'm not totally not capable. ;)

eshattow, 2022-08-14:
I have updated #5909 that shows more clearly this problem. There is no consistency in handling of units which are displayed different than their data is to be entered. Also there is some weird behavior with unary minus on parenthesis component.

adrianinsaval, 2023-04-21:
Is this still an issue with the new parser?

adrianinsaval, 2023-04-26:
this issue is mostly no longer present now that input boxes use the expression parser, and what is left can be considered a duplicate of #5909 so I'm closing it.

closed 2023-04-26 by adrianinsaval