FreeCAD/FreeCAD #6032: adding primitives changes body coordinate system

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4267

* **Reporter:** uwestoehr
* **Date submitted:** 2/15/2020
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** new
* **Tags:** 

# Original report text

- Add 2 primitives, the first one attached to the xy-plane, the second one attached to a face of the first one. Attached is an example file.
- now try to mirror the first one along the xy-plane

result: you get the error

```<App> Document.cpp(3464): Failed to recompute Cube#Mirrored: Transformation failed
Recompute failed! Please check report view.```

The problem is hereby that the addition of the primitive changes the body coordinate system. Therefore the mirroring is correctly done, but the xy-plane was moved.

# Additional information

forum thread: https://forum.freecadweb.org/viewtopic.php?f=3&t=43395

# FreeCAD Info

```
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.19603 (Git)
Build type: Release
Branch: master
Hash: 62251e3b398ce7d46aba6c84d68da990bfb8b0f9
Python version: 3.6.8
Qt version: 5.12.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: German/Germany (de_DE)

```

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by openBrain 2020-04-07 16:37
uwestoehr : I changed "global coordinate system" to "body coordinate system" as the GCS roughly can't be altered. However, do you have some evidence that the behavior is linked with coordinate system tampering ? Or is it just rough inference ? I see nothing about it in the forum discussion. ;) Thx.

adrianinsaval, 2022-03-13:
@donovaly can you still reproduce? I can mirror without issues after editing the mirror and selecting the XY plane again
[Cube.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/8240595/Cube.FCStd.zip)
(it's not re-zipped, just remove .zip extension)

donovaly, 2022-03-13:
The mirroring works now but the main issue remains: adding primitives changes body coordinate system
In the example file, add a new sketch to a face of the box and you can see that its coordinate system cener is not the one of the Part's coordinate system, bit the center of the prism.

adrianinsaval, 2022-03-13:
I don't understand, when you create a sketch on a face it gets attached to that face and the origin of the sketch is determined by it's attachment. That's normal behavior AFAIK

adrianinsaval, 2022-03-13:
Ok it appears what I said it's not true, when creating a similar shape through a pad and sketching on a face the origin is indeed the same as the body origin. Looking at the placement of the sketch it is different. For some reason attaching to a pad is different to attaching to a cube. That's the real issue IMO.

donovaly, 2022-03-13:
> For some reason attaching to a pad is different to attaching to a cube. That's the real issue IMO.

The real issue is that adding a primitive changes the coordinate system. Take this example:
[Cube.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/8240658/Cube.FCStd.zip)

Note that the origin of the sketch in the 2 bodies is different. Only the addition of the primitive is the difference. You could have attached the sketch to a pad and you would get the same problem. Therefore in practice I often add set a subtractive sphere ad the body's origin to rest the origin for new sketches.

------------------

Another issue: in my example file, change the placement of the prism in the property editor.
Result: the box is moved too in the preview, first when you leave the property editor, you get the right result.

adrianinsaval, 2022-03-13:
The body origin object is still the same, if you attach a sketch to a base plane the origin is in the correct place. It appears the placement property of the primitives is problematic because it seems this is taken into account when attaching a sketch. Pads following the insertion of a primitive have the same placement as the primitive while normally it has placement of 0
@wwmayer is there some way of uncoupling the placement of PD primitives to the placement of the shape property? IMO it should be coupled to the position of the corresponding addsubshape only while the real placement of the shape should always be 0 as with other PD features.

The second issue you mention IMO is because in the preview the shape of the object is moved but when recomputing only the addsubshape is actually moved. This is not trivial to solve though because you would need to recompute at every step if you wanted a realistic preview

wwmayer, 2022-03-14:
As far as I understand the issue the whole point is that unlike what Uwe has expected the two circles of the two sketches are not coincident.

When attaching a sketch to a face it uses the face's local coordinate system (expressed as a placement) and this is affected by the Body geometry and the OCC modeling algorithms. The placement is computed inside AttachExtension::positionBySupport() that delegates it to AttachEnginePlane and that further delegates it to AttachEngine3D::calculateAttachedPlacement.

Because the two bodies look quite different it's no wonder why the sketches' placements are different.

adrianinsaval, 2022-03-14:
The thing is that only primitives change the coordinate system of the body shape, pads and other operations don't. This is a confusing behaviour. And IMO it's conceptually wrong, because when we set a placement for a primitive we only want to move it's addsubshape, not change the shape coordinate system.

wwmayer, 2022-03-14:
> The thing is that only primitives change the coordinate system of the body shape, pads and other operations don't. 

I don't know why that is but it could be related to the internal working of OCC. When you create a standard box and determine the placement of the top face it's always empty (i.e. no offset and no rotation). However, if you create a square sketch and pad it then the top face always has an offset.

donovaly, 2022-03-15:
> As far as I understand the issue the whole point is that unlike what Uwe has expected the two circles of the two sketches are not coincident.

No, my point is the position of the origin. Sketch2 in my example is attached to the box surface. Therefore the sketch origin is at one corner of the cube. For Sketch, the same was dine, and even that the prism is after the tip and should therefore not have an effect, the origin of Sketch1 is at the center of the prism.

An important aspect is that the issue is independent on the face one attaches the sketch to. Here is a further example:
[Cube-Pad.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/8249623/Cube-Pad.FCStd.zip)
where I attached Sketch11 and Sketch 22 to the surface of a Pad. Note also in this case, and despite the prism is behind the tip and behind the sketch, it changes the origin of the sketch.

I went crazy when I first discovered it because I usually refer distances in sketches to the origin, but then I move the prism, also the origin moves and thus my further construction.

adrianinsaval, 2022-03-17:
I think the root cause of #6238 is the same as this one. There's something weird going on with primitives placement.

FlachyJoe, 2024-10-07:
I've traced the problem to `AttachEngine3D::_calculateAttachedPlacement`
https://github.com/FreeCAD/FreeCAD/blob/d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c/src/Mod/Part/App/Attacher.cpp#L1403-L1405

`refOrg` is null when the sketch is attached to a Pad face but equals the primitive or binder placement in these cases.
The sketch is at origin for pads because Pad always has a null placement, it is its profile which locate it.

We could change this by using `refOrg = (0,0,0)` in all cases but it will break models made with the current logic.

**Moreover if you attach a primitive on a primitive the current behavior is what is needed.**
You won't do this without AttachementOffset calculation if face origin is in front of (0,0,0)
[image]

closed 2025-01-30 by github-actions[bot]

luzpaz, 2025-02-01:
Can the bot check if there's an open PR for the ticket ?

maxwxyz, 2025-02-01:
I don't think so but will check next week

maxwxyz, 2025-02-03:
not possible right now