FreeCAD/FreeCAD #6040: Dimension values are displayed from reverse

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=4292

* **Reporter:** chrisb
* **Date submitted:** 3/20/2020
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** assigned
* **Tags:** 

# Original report text

See the file attached to [forum discussion](https://forum.freecadweb.org/viewtopic.php?p=378726#p378726). Double clicking on "Hole Base" shows the numbers from behind.
This vanishes as soon as something is changed in the sketch, e.g. a dimension is added. But will reoccur on close/reopen

Note that the sketches do not belong to the body.
The phenomenon vanishes if the loft is deleted and the sketches are shown in the tree on top level again.

# FreeCAD Info

```
OS: macOS High Sierra (10.13)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.20052 (Git)
Build type: Release
Branch: master
Hash: 6291921e9abe737e4440ff5814ca58547defadef
Python version: 3.8.2
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: C/Default (C)

```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by chrisb 2020-03-25 06:14
Here is [another one](https://forum.freecadweb.org/viewtopic.php?p=380205#p380200). It is again a sketch used for a feature inside of the body, while the sketch lives outside.

---

### Comment by leoheck 2020-03-25 16:12
Here is a second file with the same issue.
https://forum.freecadweb.org/viewtopic.php?p=380205#p380152

The issue does not happen in 0.18.4 but does happen in 0.19.

OS: Ubuntu 19.10 (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.
Build type: Release
Python version: 3.7.3
Qt version: 5.12.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/United States (en_US)

---

### Comment by leoheck 2020-03-25 16:18
The Sketch is inside the Body at least looking from the Model view.

The Dependency graph tells a different story. The revolution is inside the Body and the Sketch is outside of it (does not represent what the Model menu shows)

However, version FreeCad 0.18.4 interprets the Sketch correctly and the 0.19 shows the values mirrored.

luzpaz, 2022-04-01:
Bumped [forum thread](https://forum.freecadweb.org/viewtopic.php?p=585317#p585317)

PLChris, 2022-04-03:
Retested and error doesn't occur anymore.

closed 2022-04-03 by PLChris